sh123/esp32_loraprs #2: boards?

jpontius, 2020-05-06:
Hello!  This looks awesome. I would like to try it out, are there specific boards this is compatable with?  For instance could I use it with the [TTGO T-Beam ESP32](https://www.aliexpress.com/item/4000897355282.html?spm=a2g0o.productlist.0.0.4a0b2180qLUXez&algo_pvid=63b4d923-eda9-4a50-bea8-280063376781&algo_expid=63b4d923-eda9-4a50-bea8-280063376781-2&btsid=0ab6fb8815887725992247582e1afb&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)?

sh123, 2020-05-06:
Not sure, using esp32-wroom, like this one https://aliexpress.com/item/32799057508.htm, if it is compatible with TTGO T-Beam ESP32 then it should work.

jpontius, 2020-05-06:
Perfect, thanks!

closed 2020-05-06 by jpontius

sh123, 2020-05-06:
I will update readme, thanks for pointing out.