sh123/esp32_loraprs #4: This a great idea!

Cloolalang, 2020-06-11:
I really like this idea.
Im building a testing one today here in Cape Town.!
Patrick 
ZS1dBm

closed 2020-06-11 by Cloolalang

sh123, 2020-06-11:
That's great, let me know if you have any questions. On large volume of traffic may need to optimize APRS server connectivity logic to keep connection open. Currently it reconnects on each new packet.

Cloolalang, 2020-06-11:
Hi, Yes Its up and running now!
You can look me up on APRS-FI.
Im going to try to get my local club to get involved.
Thank-you for your good work, its a great contribution!.
73
ZS1dBm
Patrick

Cloolalang, 2020-06-11:
I have my gateway on the 5th Floor apartment with 7 element Yagi.
Going to drive around a little with this tracker.

Im wondering it narrower bandwidth operation would be more spectrum-friendly.
My plan is to build and test a 7.8KHz band gateway and tracker using the SX126x TCXO modules.
Is there a forum in English regarding this new mode of APRS?
73,
Patrick

sh123, 2020-06-11:
Narrow band modes might be problematic without TCXO. For experiments you can set LORAPRS_FREQ_CORR to true for the server side. This way server will tune to client's frequency based on frequency error value. This way 20.6 KHz with SF 9 worked fine. Also, check spectrum for the ISM band, if there is a lot of interference then try to tune lower or use horizontal polarization.

Cloolalang, 2020-06-11:
Yes, 
Have a look at this:
https://www.loratracker.uk/sx1262-improved-lora-device/

p

sh123, 2020-06-11:
Also, check that client's antenna is close to SWR 1, there is a lot of commercial UHF antennas with higher SWR, I was experimenting with different antennas and home made matched UHF 1/2 wave loop/halo was giving best results, UHF vertical antennas from handheld antennas did not perform well, 20 km was the furthest range over the sea.

Cloolalang, 2020-06-11:
Thanks very interesting.
One further question.
 My gateway is only RF to IS, my question; IS-to-RF possible to so its a full two-way Igate?
P

sh123, 2020-06-11:
Currently standalone server is only RF to IS, it is possible to implement IS to RF plus repeater functionality, but you can always take another client and connect it to raspberry or PC by using rfcomm and run any existing APRS software on it.

Cloolalang, 2020-06-12:
Thanks so much for answering all my questions!
Im going to make a few ESP32+ RA02/FRM96 boxes up and test my coverage with this IGate.
I have joined the APRS über LoRa group as well.
Happy summer, (Its a cold winters day here in Cape Town!).
73
Patrick
ZS1dBm

sh123, 2020-06-12:
I'll try to add support for IS->RF and repeater functionality, it would interesting to have pager style chat option over the APRS-IS, made some refactoring to make it easier.

Cloolalang, 2020-06-12:
Great. I am going design a basic "motherboard" for the ESP32 Dev module and RA-02 and RFM96 carrier PCB modules. I did notice that USB power for both modules is not adequate and so I will include a better 3.3V regulator and capacitors.
Also thinking of having external 2.5GHz antenna connector on the ESP32, (UfL).
Best Wishes
Patrick

sh123, 2020-06-13:
I've added option for IS to RF traffic gating. Now can enable it with `cfg.EnableIsToRf` option and point to some arbitrary circle with `cfg.AprsFitler` with e.g. r/xx/xx/25 radius. Quite interesting to see messages from arbitrary world region popping up on APRSDroid and visible on the map.

sh123, 2020-06-14:
Added support for basic `WIDEn-n`  packet digirepeating and own station beacon periodic announcement.

Cloolalang, 2020-06-15:
Wow! quick work!
Thanks for letting me know, Im going to test it this week.
73
ZS1DBM
Patrick

Cloolalang, 2020-06-15:
I have your latest up and running, Very good to have TX and messaging, I reduced my radius to control the time-on-air of the Igate, with SF 12, its still very busy. Fantastic work,
When I have some SX1262 radios I will send you some.
Im assuming CAD is running as I see no collisions..good. would be interesting to look at the CAD parameters.
I will carry on testing here and the digipeating etc.
73
Patrick 
ZS1DBM-1

sh123, 2020-06-15:
Oh, that's great, write if you will encounter any issues or some features will be needed.

Not sure about collision detection, the library version of https://github.com/sandeepmistry/arduino-LoRa does not seem to have any method to enable it, there is fork from https://github.com/szotsaki/arduino-LoRa and it has call to cad() to explicitly enable it.

Cloolalang, 2020-06-16:
Hi, Thanks for your response.
OK I will take a look, Im also woking with RadioHead and Mysensors which have CAD and a CSMA routine.
Here's my setup information:

https://sites.google.com/view/d34dc3ed/home

73