sh123/esp32_loraprs #7: Syncword 0xf3

Cloolalang, 2020-06-17:
I had some feedback from the TTN forum regarding my APRS IGate setup:

https://www.thethingsnetwork.org/forum/t/the-amateur-radio-corner/5252/68

Opinion on syncword.

sh123, 2020-06-17:
Could be changed, which value is suggested? Is there any common agreement in LoRa community on which one to use to be complatible?

sh123, 2020-06-17:
I think it should be 0x3f, but no 0xf3, probably typo, I'll fix that

sh123, 2020-06-17:
Changed to 0x3f

closed 2020-06-17 by sh123

Cloolalang, 2020-06-18:
Hey, thanks, I will update my setup.
I have joined this group
https://github.com/sh123/esp32_loraprs/issues/7
and there is a discussion about sync word here:
https://www.thethingsnetwork.org/forum/t/the-amateur-radio-corner/5252/70

73
Patrick
ZS1dbm

sh123, 2020-06-18:
There are more thoughts about compatibility. This project is using `AX25` when sending data over LoRa (as per http://www.aprs.org/doc/APRS101.PDF page 12) and encapsulates `AX25` frames into `KISS` when packets are transferred to/from `TNC` over serial Bluetooth to be compatible with existing `TNC KISS` interface in Linux software. However, some existing LoRa APRS software implementations just send plain APRS text messages over the RF. These formats won't be compatible.

Cloolalang, 2020-06-19:
Interesting, 
After reading this https://github.com/sh123/esp32_loraprs/issues/7#event-3452266887
I have decided to just use sync word 0x34 for now.

I am also looking at testing a non-ham ISM compatible version with CBAPRS.de

Here is another ESP32/LoRa-APRS:

https://github.com/peterus/LoRa_APRS_iGate

My IGate is working very well., I have had 17km on my first drive test, 
73
Patrick 
ZS1dBm

sh123, 2020-06-19:
Quite good range for QRPP, I'll change syncword to 0x34.