sh123/esp32_loraprs #8: CSMA and CAD mode

Cloolalang, 2020-06-19:
Further to our other chats,. I see that there is no collision control right now.
I see that KISS has P-persistent CSMA  as part of the protocol.
And the Mysensors library I use RFM95_sendWithRetry with some form of CSMA.
I think that using APRS on these Lora units is really cool, however the time-on-air at large spreading factors is quite high. 
What do you think of implementing CAD/CSMA to take advantage of the fact that we are generally using transceivers for our Lora APRS so we can have better control of the traffic flows?
73
Patrick
ZS1DBM

sh123, 2020-06-19:
Good idea, will investigate about it.

Cloolalang, 2020-06-19:
I see APRS has its own message retry feature, Im not sure if it relies on CAD or a receiver at all.
Perhaps im incorrect regarding the use of CSMA with APRS, and its only an AX25 connected mode thing.
https://unsigned.io/p-persistent-csma/

sh123, 2020-06-19:
Perhaps it would be easier to enable cad  in LoRa itself.
```
void LoRaClass::cad()
{
  idle();
  writeRegister(REG_OP_MODE, LORA_MODE_LONG_RANGE_MODE | LORA_MODE_CAD);
  delayMicroseconds(120); // IDLE -> CAD takes about ~120 µs
}
```

sh123, 2020-06-20:
I've added similar logic to https://unsigned.io/p-persistent-csma/ and updated readme, unfortunately `cad()` is not provided by the LoRa lib, so I'm just relying on `Lora::parsePacket` for that matter. Seems to work, but did not test or large volumes of traffic.

closed 2020-06-20 by sh123

Cloolalang, 2020-06-21:
Quick work friend!
I will give it a test this week. I am looking at way to co-ordinate this project.
I see that iot4pi is using coding rate 1.
It will be great to look at organising a standard some how. I will ask the group.
I got a tip from Dan Frey,  https://github.com/faydr/QMesh
He is using the EByte E22 400M30S modules. I have ordered some to make our APRS LoRa Prototypes with the SX126x radios.
73
Patrick
ZS1dBm

sh123, 2020-06-21:
Probably, it won't be easy to harmonize BW/SF/CR values as it will depend on the traffic, band plan, igate elevation and so on.

Cloolalang, 2020-06-21:
Im also looking at a non-Ham version using cbaprs.de and packet-radio.nl.
Im going to setup a 869MHz IGate and tracker, ISM compliant linked to one of these servers. It would be interesting to offer APRS to those who don't have a HAM license. I guess careful selection of frequency and duty cycle. 
P

sh123, 2020-06-22:
I have some plans to experiment with more broadband 2.4 GHz LoRa modules, such as E28-2G4M27S, more wide band modes could be used at 2.4 GHz. They offer 0.595-2 Mbps, may be good addition to https://hamnetdb.net for point-to-point line of sight links.