sh123/esp32_loraprs #9: Distance record South Africa today

Cloolalang, 2020-07-11:
Just an update to let you know that I have a new distance record with my gateway to mobile 29,97KM in Cape Town.
https://sites.google.com/view/d34dc3ed/home
73!
Patrick
ZS1dbm

sh123, 2020-07-11:
Good for QRPP with some tens of milliwatts, have fiberglass mast, was thinking to mount modem on the top to increase elevation, from signal report it seems there is still some room.

Cloolalang, 2020-07-17:
Yeah, interesting idea, I have fitted external WIFI antennas on ESp8266 before, it does really improve the range, so possibly you could do that for your ESP32.
This week Im trying 7,8KHz BW, SF7, Coding 5.
The modules need to be tuned up together, (I use an RTL-SDR  which I "calibrate" to my local DVB-T TV transmitter signal)
At 25C one radio is 1.6KHz too high, the other about 1.4KHz too low, so I have tuned them to the 433.7750 according to the HDSDR software.
They seem stable once tuned, they drift a little but not more than 300Hz in the lab.
Time-on-air is comparible to AX25 on FM with SF7, Coding rate 5.
I did try SF11 and Coding rate 7, but I saw a few collisions...I did change CfgPollDelayMs : to 1500 to reduce collsions it that correct?
Anyway, I will go out and test it and see, if I like the "narrow modes". I will see about fitting a temperature sensor and writing a temperature correction routine to keep the radio on frequency.
Cheers
P

sh123, 2020-07-17:
> I did try SF11 and Coding rate 7, but I saw a few collisions...I did change CfgPollDelayMs : to 1500 to reduce collsions it that correct?
You can decrease `CfgCsmaProbBoundary` this will cause larger probabilistic "waiting" before transmitting.

Cloolalang, 2020-07-18:
Arr, ok thank-you.
I am seeing that a lot of IS-RF packets from my IGate are not being shown on my APRS droid. Im not sure whats happening,
Is there a way see packets decoded success/fail on the Tracker serial output?
73
Patrick

Cloolalang, 2020-07-18:
I just put this into the sketch.

///////////////////////////////////////////////////////

 int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");
    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());}
/////////////////////////////////////////////////////////

it seems to work
Cheers

sh123, 2020-07-18:
Yes, you can connect server using USB and open console, it should display incoming packets.

closed 2020-10-18 by sh123

sh123, 2020-12-07:
@Cloolalang, I've build small Android application, which enables DV communication by using Codec2 over lora esp modem, it works similar to APRSDroid, but instead of ax25 packets it sends codec2 encoded speech from the mic and back to the speaker, https://github.com/sh123/codec2_talkie