sh123/esp32_loraprs #11: Possibility to adapt code for the T-Beam ?

TomF8COD, 2021-01-31:
Hello, 
I'm trying to figure out how to adapt the code to the ttgo tbeam as it seems a nice device for aprs use.
I've disabled the aprsIS and wifi part to be able to fit the code in the tbeam, the idea is to only use it as a kiss radio modem, without gateway feature for now. As the gateway from lora-aprs on github is working good, 
It pairs with aprsdroid and send a lora packet as far as I can see on the sdr. But the lora32-v2 using the lora-aprs/LoRa_APRS_iGate code doesn't decode the packet (it does when I'm using OE5BPA tracker code). 
As I am a total noob in development, I thought I should ask if you have any idea or suggestion I could try to make the code run on the t-beam. This device and aprsdroid would be a good match for a very nice aprs package if it can be adapted.
Thanks for any help and for the work you already did !
73 de Tom F8COD

sh123, 2021-02-01:
Hi, Tom, good to check if it uses the same 0x34 LoRa sync word also I see that LoRa_APRS_IGate does not use AX.25 UI packets, it sends raw text frames, see https://github.com/sh123/esp32_loraprs#protocol-compatibility

TomF8COD, 2021-02-01:
Thank you for your answer !
I agree for AX.25 it's the good way to go. I'll check the sync word on the iGate, I'll keep you posted if I can make it work.

73 !

sh123, 2021-02-01:
I can add dynamic packet type detection and convert it from text to ax25 and back.

TomF8COD, 2021-02-01:
Don't worry, as I'm not sure of what I'm doing, I don't want to give you
more work HI! I'm going to see if I can make another t-beam into a working
kiss modem to see if I can get the two of them communicating, else I'll get
a esp32 with more room to fit your complete code 😊

TomF8COD, 2021-02-01:
Don't worry, as I'm not sure of what I'm doing yet I don't want to give you more work :smile:  
What I'm thinking about right now, is to put the simplified code on another t-beam that I could use as a kiss modem on a linux computer. So I'll run the gateway on the computer itself (with xastir), and see if I can exchange packets between aprsdroid with t-beam1 on one side and linux with t-beam2 on the other side. 
It would make a "lightweight" version of the code that could run on small platforms, making a minimalist tracker with aprsdroid and the same "modem" could be use as a gateway on a small computer or raspberry. I don't know if it's a good idea, It's just something that came to my mind while talking about it with other hams that are getting their feet in lora aprs. 
The point that I like in your code is the possibility to send messages to others stations, this is what is missing from the simple tracker codes I've tried, and if I can recall this rely on AX.25 ?

sh123, 2021-02-01:
If you need very lightweight solution without using esp32, then check https://github.com/sh123/nrf24l01_arduino_kiss_modem, this is lightweight KISS modem, it uses (Arduino Beetle](https://www.dfrobot.com/product-1075.html) and can operate through USB (including USB OTG on Android with APRSDroid), you will need similar approach, but use LoRa library instead of nrf24l01, just make a branch and modify for LoRa usage.

Linux software is using AX.25 for APRS over KISS, so I doubt it will be interoperable with LoRa_APRS_iGate if they wont' add support for AX.25.

TomF8COD, 2021-02-01:
Oh I didn't see it! Of course, it will be easier like that, I'm going to
take a look at it right now. Thank you very much!

TomF8COD, 2021-02-01:
In the meantime, I've put the code I modified on another t-beam with another phone and I can make the two aprsdroid talk together, they see each other's position and can exchange messages. I'm happy :+1: 
I'll try to work with the other code too, it's a good way to learn :wink:

sh123, 2021-02-01:
Oh, good to hear, also, try to run https://github.com/sh123/codec2_talkie for voice communication :)

TomF8COD, 2021-02-01:
It's on my shopping list :) Thanks for all this work. Do you want me to
close the issue or do you do it ?

closed 2021-02-01 by sh123

sh123, 2021-02-03:
@TomF8COD, what was the change to adopt it for T-Beam usage so I can update README with instructions?

TomF8COD, 2021-02-03:
Hi,
I've made a fork and a branch "no_wifi" here :
https://github.com/RadioHome/esp32_loraprs

Basicaly, I've stripped everything for the wifi, the beacon, the
digipeater. As the purpose is to use it as a simple kiss tnc I figured I'll
be able to use this feautures on the applications that are using the tnc.
I've been able to test it with xastir (kiss tnc, not as ax25 tnc yet)
configured as an iGate, it works perfectly.

I'm going to do some tests with the coding rate, it seems that I loose the
link between my tracker and my base sooner with CR7 than with CR5. If I'm
right, the transmission is a little bit longer with cr7 as it encodes on 7
bits, I don't know if it's just a one time glitch, or if it is critical for
mobile use.
I've seen that Fossasat-1 is using CR5 too, so I thought if it's good for a
satellite, it's good for my wheelchair or my car ^^

sh123, 2021-02-03:
@TomF8COD , I've added single sketch arduino tnc kiss modem, adopted from nrf sketch, https://github.com/sh123/lora_arduino_kiss_modem

TomF8COD, 2021-02-03:
Awesome ! Thank you very much ! 
I was thinking about doing it, but I'm not efficient as you are, I have no coding experience, I'm more a tinkerer ;-)

sh123, 2021-02-03:
@TomF8COD , I do not have hardware yet to test sketch, was thinking about very small modem, which plugs directly into the Android USB OTG (kind of phone extension with 3d printed case) and use it with APRSDroid for tracking/messaging and DV communication with codec2_talkie.

TomF8COD, 2021-02-03:
That's a good idea, I was just talking with a friend a few minutes ago, he
has these lora modules :
https://www.zigbeesolutions.com/lora/ebyte-e32-433t30d-sx1278-sx1276-433mhz-109121318425631437.html
with 30dbm output it would be interesting for DV application with a bit
more power. We're going to try to use them.
By the way, I have a question : For a device that could be used like this
in aprs and DV, would there be a way to setup the lora modulation
parameters (BW, SF, CR) externaly,
without the need to re-flash the device ?
I haven't tested DV yet, I'll try to do it this afternoon.

sh123, 2021-02-03:
Yes, planning to add support for LoRa parameters change in esp32_loraprs modem and codec2_talkie Preferences through KISS protocol extended commands, also get signal report, so UI can show signal level as value or on S meter.

TomF8COD, 2021-02-03:
Really cool !

TomF8COD, 2021-02-06:
@sh123 I've installed xastir on a raspberry pi 3 with bluetooth connection to the t-beam at boot by adding rfcomm bind 0 xx:xx:xx:xx in a systemd service file, it's running nice. Xastir have the kiss tnc under /dev/rfcomm0 and a internet server connection configured, it works as a full time igate now.
I was wondering, would it be possible to implement the wifi use by a variable in the code, that could force the server mode, digi, beacon off with some #ifdef ? If you think it's worth to do, I can work on it but I'm afraid you will have to review my code as it may be sloppy, so I'll totally understand if you don't want, . The goal is to keep with the other changes you are implementing easily. 
I'm going to work on the lora-arduino-kiss-modem, it's a nice sketch to learn.

sh123, 2021-02-06:
@TomF8COD , if you select client mode then wifi and other logic is disabled, no need for #ifdef-s.

TomF8COD, 2021-02-06:
Ok I get how it works now. I really appreciate the time you're taking to explain me your code, it's nice. :+1:

TomF8COD, 2021-02-06:
@sh123 I've tried the client mode, but it's too big for the t-beam if I keep the #include <WiFi.h>. Don't worry, I'll try to find a solution, it's a good exercise. :wink: 
Oh I forgot one info : The t-beam integrated led is on pin 4

sh123, 2021-02-06:
@TomF8COD , select different partition scheme: "Minimal SPIFFS" in Arduino IDE Tools.

TomF8COD, 2021-02-06:
@sh123 I was wondering where this setting was, I only see it when selecting "ESP32 Dev module" but not when I choose "t-beam" or "lora32". It's flashing, but on the serial monitor I get this : 
LoRa init: 433775000, 125000, 12, 5, 20, 34, 0.........................
So it doesn't work. I'm going to investigate it tomorrow.

sh123, 2021-02-07:
@TomF8COD , seems that LoRa cannot start for some reason (pinouts?), there is no partition layout option when T-Beam is selected. Unfortunately, I do not have T-Beam to test.

TomF8COD, 2021-02-07:
@sh123 Well done, I found your answer in my mail this morning as I started the computer, I had the idea to check the boards.txt to try something like that. It works perfectly !
I tried it in client mode on the t-beam, and in server mode on a ttgo-lora32-V2 running by itseland not connectedf,  it seems to be ok.
For the integrated led pin, I think just a declaration in config.h would be enough end indicating in comment the pin 4 for t-beam and pin 25 for ttgo-lora32-v2. What do you think about it ?
If you want me to run some tests with specific settings on theses devices tell me, I have time to do it and I'll be happy to help.

And if I may ask, what is your name ? 

[image]

sh123, 2021-02-07:
@TomF8COD , changed config so that it will use lora pinout definitions and builtin led definition taken from arduino's board variant `pins_arduino.h`, this way it is just enough to select board in arduino and build without changing code.

TomF8COD, 2021-02-07:
Very good, it's ways simpler like this, indeed.

TomF8COD, 2021-02-10:
I'm starting to get a grip on git and I've worked my way into making a new branch to integrate a ssd1306 oled to display packets. Not of much use on the client, but I had already the display on my devices and it was a good exercise to make it work. 
On the standalone gateway it's more useful, especially during tests as we can see what's going on.
I used the Adafruit ssd1306 library, but I've seen the u8g2 lib that seems good too.
I've tried to activate is-to-rf on the server, but it's really heavy duty on transmitting time. It would be interesting to be able to route only messages from is to rf, I'm going to read the protocol spec to see how it works.

sh123, 2021-02-11:
@TomF8COD , you can just specify multiple filters, they just need to be space separated, for example 

```
r/lat/lon/dist t/m
```
will accept all messages packets from the circle with radius dist at point lat/lon

http://www.aprs-is.net/javAPRSFilter.aspx

TomF8COD, 2021-02-11:
@sh123 I didn't see the t filter, it's exactly what I need.
The code with the oled branch is here : https://github.com/RadioHome/esp32_loraprs
It displays the packet, if it's RX or TX and where from (NET, BCN, iGate or Digi). And of course at boot it displays some infos about the configuration.

sh123, 2021-02-12:
@TomF8COD , we can merge it, maybe just need some config parameter to disable/enable screen if board does not have it and config parameters to change pinouts.

TomF8COD, 2021-02-12:
@sh123 I'm going to watch the config and to switch from the adafruit-ssd1306 lib to u8g2 lib that manages more screens, including the ssd1306. I try to do it soon and I come back to you ;-)

TomF8COD, 2021-03-03:
@sh123 I've tried the u8g2 lib, but it lacks word wrap, so I haven't been able to get a clean display of the packets on the oled, I went back to ssd1306 for now. 
I noticed something with the digipeater, I see in ax25_payload.cpp that you insert the igate callsign defined by ownCallsign in the ax25 packet, but it doesn't seem to work. Here is what I get when I send a packet with the client tnc from F8COD-7. It should add the server callsign (my server is F8COD-10), even if it's the same call with different SID, right ? 
[image]

sh123, 2021-03-04:
@TomF8COD , it should insert own callsign only when TRACE is used, for WIDE it is not inserted.

TomF8COD, 2021-03-18:
@sh123 I've worked on the code to display infos with a 0.96" SSD1306 based Oled, the tnc displays some information on start up, then it displays the received packets and when it switches to TX. If you want to look at the code it's here : https://github.com/RadioHome/esp32_loraprs
I've installed the tnc on a raspberry-pi that already has my AEA PK12 for APRS and setup aprx to make a bi directional gateway between 433.775 (lora) and 144.800 (aprs) We're going to have fun here testing loraprs ! 
[image]
[image]
[image]
[image]

sh123, 2021-03-18:
@TomF8COD , wow looks nice :)