sh123/esp32_loraprs #16: no TX via LoRa

emsieg, 2021-06-22:
Hi, 
may I ask for some help?

I´m not able to TX/RX any APRS packets via antenna.
I see, the board is connected via WiFi to APRS Server and is sending beacons to the server.
But, no RX or TX via antenna is possible.

I´m using a Heltec WiFi LoRa32 V2 board and defined
#define LORA_RST              14
#define LORA_IRQ              26
#define SS               18 (I have added it later as I thought it could be the issue as no definition is shown)

I put my other iGate next to it,   testing any TX
and also my Tracker ,                  testing any RX.

I have installed different SW and see with that SW, 
I´m able to RX/TX any LoRa packets, so HW is not defect.

and clue

regards

sh123, 2021-06-23:
@emsieg , if you can see packets being transmitted (for example, check with SDR radio) then there is a problem with decoding, if packets are not transmitted then there is a problem with pinout definition.

TomF8COD, 2021-06-23:
Hi @emsieg be aware that this FW uses ax25 and kiss frames, so it won't be
decoded by other igate like the one from github/lora-aprs and the sh123 tnc
in server mode won't decode lora-aprs tracker.
Also be sure to test with
#define CFG_LORA_USE_ISR false in config.h line 22, I had some issue with
it when using the ttgo lora32.
As sh123 suggested, do you have a sdr to check if your lora32 is sending
anything ?

sh123, 2021-06-23:
If packet was received, but ax.25 decoding failed then `Skipping non-AX25 payload` log line should be visible in the console log. It is possible to send such packet as is to APRS assuming that it is text based submission, so that incoming packets from text based trackers could be gated.

emsieg, 2021-06-23:
Hi,
I installed a SDR SW on my laptop for testing.
here I see, that the Heltec is sending something OTA.
This packet get not been recognized on other Lora devices.

I also see that packet received on my tracker, but this heltec igate is not working with that.
[image]
[image]
[Console.txt](https://github.com/sh123/esp32_loraprs/files/6701092/Console.txt)

I have added 2 screenshots,
Heltec iGate -12 is sending, 
Tracker -11         is sending 
and Console

thanks for helping

sh123, 2021-06-23:
@emsieg , looks like LoRa module is initialized and packet is sent out. What are other devices where you are trying to receive this packet?

emsieg, 2021-06-23:
Hi,
-11 is my tracker, which is only TX
-10 is my other iGate which is only RX
-12 is the new iGate with digipeat enabled, this now device)

I expect to see -12(this Lora) to be seen in display on -10 during -12 TX; just like -11(tracker).
when the tracker (-11) is sending, i expect it to be seen it in display on -12.

I powered-off -10(iGate) to see, if -11(tracker) is forwared to APRS.fi, but did not.

it looks to me as i send something on -12, which in incompatible to the other Lora devices.

sh123, 2021-06-23:
@emsieg , what are other LoRa devices? Which firmware is used there?

closed 2021-06-25 by sh123