sh123/esp32_loraprs #17: Feature request: stripdown version of the sketch to use as walkie-talkie

happytm, 2021-07-30:
I have installed your great app https://github.com/sh123/codec2_talkie.

can you make simple one file sketch for ESP32 to use with above app removing all Kiss, APRS and WiFi Stuff for general purpose (non HAM) use ?

I am not well versed into HAM radio stuff. If there is simple sketch dealing with codec2 raw packets with LoRa transport my plan is to use it as a template to implement some newer LoRa chips like SX1262 and SX1280.

Thanks.

sh123, 2021-07-30:
There is a single sketch project for Arduino https://github.com/sh123/lora_arduino_kiss_modem

happytm, 2021-07-30:
That sketch looks very good and simple. Unfortunately I wanted to use Bluetooth link between modem and phone on both side of walkie-talkie.

Thanks.

sh123, 2021-07-30:
@happytm in client mode this modem does not use aprs and wifi, but kiss still a requirement.

happytm, 2021-07-30:
OK. Understood . Thanks.

closed 2021-07-30 by sh123