sh123/esp32_loraprs #19: Question: Interfacing Codec2 DV with a different RF library/device like nrf24

RasaKh, 2021-09-11:
Hi there.
Would it be possible to interface the Android Codec2 talkie app with your arduino code with different RF modules to send the raw audio packets through?
If so which parts of your code (which libraries) would have to be used or modified?
I do realize this question is not directly related to your project but I think this should be possible so any hint is highly appreciated.

sh123, 2021-09-11:
You need to modify places where communication with LoRa module is done and use your own module routines there.

closed 2021-09-11 by sh123

sh123, 2022-12-14:
As project is using RadioLib now, it should be doable with minimal modifications https://jgromes.github.io/RadioLib/classn_r_f24.html