sh123/esp32_loraprs #20: Add support to S126X.

GabeHC, 2021-09-29:
Can you add support to S126X Lora chips?   
I been trying to change Lora library without succeed.

sh123, 2021-09-30:
@GabeHC , need to completely switch over to RadioLib, current LoRa library, which is in use, supports only some subset of chipsets and not actively maintained.

mrpatrick1991, 2021-09-30:
I'll try to put together a PR. RadioLib can also do some other cool things, too.

sh123, 2021-10-16:
@GabeHC , @mrpatrick1991 , I've added preliminary support for RadioLib into https://github.com/sh123/esp32_loraprs/tree/radiolib branch, will test it later

sh123, 2021-10-17:
@mrpatrick1991 , got it working also for receive, but only for bandwidths >= 62.5 kHz, reported issue back to RadioLib project. Also, need to use ISR, `CFG_LORA_USE_ISR = true`, there is no way to poll for new data without blocking in RadioLib and when polling it misbehaves returning the same packet all the time.