sh123/esp32_loraprs #27: ESP32 hangs after getting package from lora Tracker!

GabeHC, 2021-10-31:
Hardware : [ESP32 Lora board](https://easyeda.com/GabeH/esp32-lora-aprs) design by myself ([BV5DJ](https://www.qrz.com/db/bv5dj))
Lora : E22 SX1268
config: server mode (APRS-IS gate mode)
ERR:  

> [INFO]_ sendPeriodicBeacon : Periodic beacon is sent
Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Core 1 register dump:
PC      : 0x400d57c6  PS      : 0x00060330  A0      : 0x800d2ec8  A1      : 0x3ffcfe60
A2      : 0x3ffc4a9c  A3      : 0x00000001  A4      : 0x0000003f  A5      : 0x00000000
A6      : 0x000000c0  A7      : 0x3ffcff48  A8      : 0x800d57c6  A9      : 0x3ffcff00
A10     : 0xc22c0000  A11     : 0x000000c0  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x00060b20  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000004
EXCVADDR: 0x00000000  LBEG    : 0x40092d59  LEND    : 0x40092d69  LCOUNT  : 0xfffffffe
Core 1 was running in ISR context:
EPC1    : 0x400d57c6  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40090db8
ELF file SHA256: 0000000000000000
Backtrace: 0x400d57c6:0x3ffcfe60 0x400d2ec5:0x3ffcff20 0x400d2ec5:0x3ffcff70 0x400d5acd:0x3ffcffe0 0x400d2939:0x3ffd0020 0x400e1904:0x3ffd0040 0x40094cf2:0x3ffd0060
Rebooting...
ets...

sh123, 2021-10-31:
Seems that reading in ISR does not work for sx1268 module, I'll try to find some generic solution to that.

sh123, 2021-10-31:
Please, try with this branch https://github.com/sh123/esp32_loraprs/tree/taskread, it will spawn separate reader task instead of reading in ISR

GabeHC, 2021-10-31:
Where are you located? I can get you one for testing.
I try to run it on Lilygo T-Beam V1.1. nothing ever heard on it.
not even get into ISR. what went wrong with it?

sh123, 2021-10-31:
Can you get working one of these? https://github.com/jgromes/RadioLib/tree/master/examples/SX126x

GabeHC, 2021-10-31:
Yes, It work with RadioLib examples without any ptoblem.
Dio0/Dio1,  getModemStatus(), getFrequencyError() are all fixed

sh123, 2021-10-31:
Where are your changes, can you point to the branch with your changes?

sh123, 2021-11-15:
@GabeHC , received modules, will try them out, have you got them working?

GabeHC, 2021-11-16:
unfortunately no!
I even try it on two TTGO T-Beam, still not working.  Tracker uses [this one](https://github.com/lora-aprs/LoRa_APRS_Tracker)
my code is [here](https://github.com/GabeHC/esp32_loraprs/tree/SX126X)

GabeHC, 2021-11-16:
> @GabeHC , received modules need to perform some soldering, will try them out, have you got them working or still having problems with receiving?
I suspect it is the getModemStatus() that I have simply return not busy for all cases. It is not transmitting nor receiving!

sh123, 2021-11-16:
I think `#if MODULE_NAME == SX1268` statement is not gonna work, I've made some changes to make it work in master.

GabeHC, 2021-11-18:
#ifdef MODULE_NAME == SX1268 should not be a problem, cause it is the macro we specified in module_name.h,  and yes RXEN/TXEN were set to 2,4, and Builtin LED set to 12 in my case.

Following is my config.h 

> /*#include <font6x8.h>
#include <nano_engine.h>
#include <nano_gfx.h>
#include <nano_gfx_types.h>
#include <sprite_pool.h>
#include <ssd1306.h>
#include <ssd1306_16bit.h>
#include <ssd1306_1bit.h>
#include <ssd1306_8bit.h>
#include <ssd1306_console.h>
#include <ssd1306_fonts.h>
#include <ssd1306_generic.h>
#include <ssd1306_uart.h> */
//
#define LED_TOGGLE_PERIOD     1000
//
#define SERIAL_BAUD_RATE      115200
//
// change pinouts if not defined through native board LORA_* definitions
#define USE_RADIOLIB
//
#ifndef LORA_RST
#pragma message("LoRa pin definitions are not found, redefining...")
#define LORA_RST              14
#define LORA_IRQ              33
#endif
//
#ifndef BUILTIN_LED
#pragma message("BUILDIN_LED is not found, defining as 12")
#define BUILTIN_LED           12
#endif
//#define BUILTIN_LED           12
#define CFG_IS_CLIENT_MODE    false        // false - server mode (APRS-IS gate mode)
/*
// lora pinouts, CAD and ISR usage
#define CFG_LORA_PIN_SS       SS
#define CFG_LORA_PIN_RST      LORA_RST
#define CFG_LORA_PIN_DIO0     LORA_IRQ
#ifdef USE_RADIOLIB
#define CFG_LORA_PIN_DIO1     LORA_IRQ    // set to your DIO1 pin number if connected
#define CFG_LORA_USE_ISR      true        // always ON for RadioLib
#else
#define CFG_LORA_PIN_DIO1     RADIOLIB_NC   // not used in arduino-LoRa
#define CFG_LORA_USE_ISR      false       // set to true for ISR usage in arduino-LoRa
#endif
#define CFG_LORA_USE_CAD      false        // set to true to utilize carrier detection
*/
// lora pinouts, CAD and ISR usage
#define CFG_LORA_PIN_SS       SS
#define CFG_LORA_PIN_RST      LORA_RST
#define CFG_LORA_PIN_A        LORA_IRQ    // (sx127x - dio0, sx126x/sx128x - dio1)
#ifdef USE_RADIOLIB
#define CFG_LORA_PIN_B        RADIOLIB_NC // (sx127x - dio1, sx126x/sx128x - busy)
#define CFG_LORA_USE_ISR      true        // always ON for RadioLib
#else
#define CFG_LORA_PIN_DIO1     LORA_IRQ    // not used in arduino-LoRa
#define CFG_LORA_USE_ISR      false       // set to true for ISR usage in arduino-LoRa
#endif
#define CFG_LORA_USE_CAD      false       // set to true to utilize carrier detection
//
//
// lora protocol parameters
#define CFG_LORA_FREQ         438.20E6
#define CFG_LORA_BW           125e3
#define CFG_LORA_SF           12
#define CFG_LORA_CR           8
#define CFG_LORA_PWR          22
#define CFG_LORA_ENABLE_CRC   true        // set to false for speech data
//
// wifi client and AP options
#define CFG_WIFI_ENABLE_AP    false       // run as wifi access point, for CFG_KISS_TCP_IP mode
#define CFG_WIFI_SSID         "GabeAP"      // connect to SSID or run as this SSID in AP mode
#define CFG_WIFI_KEY          "22433897"  // wifi key
//
// bluetooth
#define CFG_BT_NAME           "loraprs"
#define CFG_BT_USE_BLE        true        // set to true to use bluetooth low energy (for ios devices)
//
// USB
#define CFG_USB_SERIAL_ENABLE false       // true - enable communication over USB Serial
//
// KISS protocol options
#define CFG_KISS_EXTENSIONS   true    // true - enable modem control from application with KISS commands
#define CFG_KISS_TCP_IP       false   // true - run as KISS TCP/IP server, no bluetooth operations performed
//
// APRS-IS options, valid in when CFG_IS_CLIENT_MODE = false
#define CFG_APRS_LOGIN        "BV5DJ-10"
#define CFG_APRS_PASS         "12345"
#define CFG_APRS_FILTER	      "r/24.11/120.42/50"
//#define CFG_APRS_RAW_BKN      "BV5DJ-10>APZMDM,WIDE1-1:!0000.00N/00000.00E#LoRA 433.775MHz/BW125/SF12/CR7/0x34"
#define CFG_APRS_RAW_BKN      "BV5DJ-10>APZMDM,WIDE1-1:=2411.2 N/12041.9 ErESP32-LORAPRS TESTING QTH @ Taichung"
// APRS-IS gateway options, valid in when CFG_IS_CLIENT_MODE = false
#define CFG_PERSISTENT_APRS   false   // keep tcp/ip connection open (lot of traffic), otherwise connect on new packet (very rare traffic)
#define CFG_DIGIREPEAT        false   // digirepeat incoming packets
#define CFG_RF_TO_IS          true    // forward packets from radio to internet
#define CFG_IS_TO_RF          true    // forward packets from internet to radio basedon CFG_APRS_FILTER
#define CFG_BEACON            true    // enable perdiodc beacon from CFG_APRS_RAW_BKN
#define CFG_TEXT_PACKETS      false   // enable aprs text packets instead of binary for interoperability with other projects (disables KISS+ax.25)
//
// frequency correction for narrow band bandwidths
#define CFG_FREQ_CORR         false   // true - correct own frequency based on received packet
#define CFG_FREQ_CORR_DELTA   1000    // correct when frequency difference is larger than this value
//
// PTT control
#define CFG_PTT_ENABLE        false   // enable external ptt (relay) control (for amplifier)
#define CFG_PTT_PIN           4       // PTT pin
#define CFG_PTT_TX_DELAY_MS   50      // time between relay swited on and transmission
#define CFG_PTT_TX_TAIL_MS    10      // time between stop transmission and relay off

sh123, 2021-11-18:
I think `CFG_LORA_PIN_B` needs to be set to BUSY pin as in RadioLib examples.

GabeHC, 2021-11-19:
Now it does transition, but still nothing heard. no mater it is set to Text or Binary mode. seams ISR is not active. 

>  [INFO] printConfig : Current mode: APRS-IS iGate
[INFO] printConfig : Built with RadioLib library
[INFO] printConfig : Using TNC2 text mode
[INFO] printConfig : UsbSerialEnable: no
[INFO] printConfig : EnableSignalReport: yes
[INFO] printConfig : EnablePersistentAprsConnection: yes
[INFO] printConfig : EnableRfToIs: yes
[INFO] printConfig : EnableIsToRf: no
[INFO] printConfig : EnableRepeater: no
[INFO] printConfig : EnableBeacon: yes
[INFO] setup : KISS extensions are disabled in TNC2 mode
[INFO] setupLora : Initializing LoRa
[INFO] setupLora : Frequency: 438200000 Hz
[INFO] setupLora : Bandwidth: 125000 Hz
[INFO] setupLora : Spreading: 12
[INFO] setupLora : Coding rate: 8
[INFO] setupLora : Power: 22 dBm
[INFO] setupLora : Sync: 0x34
[INFO] setupLora : CRC: enabled
[INFO] setupLora : Using SX126X module
[INFO] setupLora : LoRa initialized
[INFO] setupWifi : WIFI connecting to GabeAP
[WARN] setupWifi : WIFI retrying 0
[WARN] setupWifi : WIFI retrying 1
[WARN] setupWifi : WIFI retrying 2
[WARN] setupWifi : WIFI retrying 3
[WARN] setupWifi : WIFI retrying 4
[INFO] setupWifi : WIFI connected to GabeAP
[INFO] setupWifi : IP address: 10.0.0.152
[INFO] setupBt : BT init loraprs
[INFO] setupBt : BT initialized
[INFO] reconnectAprsis : APRSIS connecting to rotate.aprs2.net
[INFO] reconnectAprsis : APRSIS connected
[INFO] reconnectAprsis : APRSIS logged in
[INFO] sendPeriodicBeacon : Periodic beacon is sent
[INFO] onAprsisDataAvailable : # aprsc 2.1.10-gd72a17c
[INFO] onAprsisDataAvailable : # logresp BV5DJ-10 verified, server T2ALBERTA
[INFO] onAprsisDataAvailable : # aprsc 2.1.10-gd72a17c 19 Nov 2021 10:22:17 GMT T2ALBERTA 44.184.136.10:14580
[INFO] onAprsisDataAvailable : # aprsc 2.1.10-gd72a17c 19 Nov 2021 10:22:37 GMT T2ALBERTA 44.184.136.10:14580

sh123, 2021-11-20:
Soldered module, works fine for both RX/TX when setting `CFG_LORA_USE_ISR` to `false`

GabeHC, 2021-11-21:
Still not receiving. I have no problem with SX126x_Receive_Interrupt  example.
```
// comment out for arduino-Lora usage
#define USE_RADIOLIB

// Uncomment for SX126X module usage
#define USE_SX126X

// Check your module name at https://github.com/jgromes/RadioLib/wiki/Modules
#ifdef USE_SX126X
#define MODULE_NAME   SX1268
#else
#define MODULE_NAME   SX1278
#endif

// generic options
#define LED_TOGGLE_PERIOD     1000
#define SERIAL_BAUD_RATE      115200
#define CFG_LOG_LEVEL         DebugLogLevel::LVL_INFO

// change pinouts if not defined through native board LORA_* definitions
#ifndef LORA_RST
#pragma message("LoRa pin definitions are not found, redefining...")
#define LORA_RST              14
#define LORA_IRQ              33
#endif

// redefine LED if not defined in Arduino to have module heartbeat indication
#ifdef BUILTIN_LED
#undef BUILTIN_LED
#endif
#pragma message("BUILTIN_LED defining as 12")
#define BUILTIN_LED           12

// select between client mode and APRS-IS gate mode
#define CFG_IS_CLIENT_MODE    false       // false - server mode (APRS-IS gate mode)

// lora generic pinouts
#define CFG_LORA_PIN_SS       SS
#define CFG_LORA_PIN_RST      LORA_RST
#define CFG_LORA_PIN_A        LORA_IRQ    // (sx127x - dio0, sx126x/sx128x - dio1)
#define CFG_LORA_PIN_B        39          // (sx127x - dio1, sx126x/sx128x - busy)

// lora pinouts, SX126X RXEN/TXEN usage
#ifdef USE_SX126X
#define CFG_LORA_PIN_RXEN     2           // (sx127x - unused, sx126x - RXEN pin number)
#define CFG_LORA_PIN_TXEN     4           // (sx127x - unused, sx126x - TXEN pin number)
#endif

// CAD and ISR usage selection
#ifdef USE_SX126X
#define CFG_LORA_USE_ISR      false       // reading in ISR does not work on sx126x
#define CFG_LORA_USE_CAD      true        // do not transmit if channel is busy
#else
#define CFG_LORA_USE_ISR      true        // true - read incoming data in ISR, false - do not read in ISR
#define CFG_LORA_USE_CAD      true        // set to true to utilize carrier detection
#endif
```

sh123, 2021-11-21:
IRQ pin is responsible for interrupt, maybe you have LORA_RST defined in arduino so it is not getting defined, try to set explicitly
```
#define CFG_LORA_PIN_A        33 
```

GabeHC, 2021-11-22:
That is not it. I saw "LoRa pin definitions are not found, redefining..." during compile. 
```
// lora generic pinouts
#define CFG_LORA_PIN_SS       5
#define CFG_LORA_PIN_RST      14
#define CFG_LORA_PIN_A        33          // (sx127x - dio0, sx126x/sx128x - dio1)
#define CFG_LORA_PIN_B        39          // (sx127x - dio1, sx126x/sx128x - busy)
```
also not working!
```
[INFO] printConfig : Current mode: APRS-IS iGate
[INFO] printConfig : Built with RadioLib library
[INFO] printConfig : Using TNC KISS and AX.25 mode
[INFO] printConfig : UsbSerialEnable: no
[INFO] printConfig : EnableSignalReport: yes
[INFO] printConfig : EnablePersistentAprsConnection: yes
[INFO] printConfig : EnableRfToIs: yes
[INFO] printConfig : EnableIsToRf: no
[INFO] printConfig : EnableRepeater: no
[INFO] printConfig : EnableBeacon: yes
[INFO] setupLora : Initializing LoRa
[INFO] setupLora : Frequency: 438200000 Hz
[INFO] setupLora : Bandwidth: 125000 Hz
[INFO] setupLora : Spreading: 12
[INFO] setupLora : Coding rate: 8
[INFO] setupLora : Power: 22 dBm
[INFO] setupLora : Sync: 0x34
[INFO] setupLora : CRC: enabled
[INFO] setupLora : Using SX126X module
[INFO] setupLora : LoRa initialized
[INFO] setup : Reading data on separate task
[INFO] setupWifi : WIFI connecting to GabeAP
[INFO] processIncomingDataTask : Incoming data process task started
[WARN] setupWifi : WIFI retrying 0
[WARN] setupWifi : WIFI retrying 1
[WARN] setupWifi : WIFI retrying 2
[WARN] setupWifi : WIFI retrying 3
[WARN] setupWifi : WIFI retrying 4
[INFO] setupWifi : WIFI connected to GabeAP
[INFO] setupWifi : IP address: 10.0.0.152
[INFO] setupBt : BT init loraprs
[INFO] setupBt : BT initialized
[INFO] reconnectAprsis : APRSIS connecting to rotate.aprs2.net
[INFO] reconnectAprsis : APRSIS connected
[INFO] reconnectAprsis : APRSIS logged in
[INFO] sendPeriodicBeacon : Periodic beacon is sent
[INFO] onAprsisDataAvailable : # aprsc 2.1.10-gd72a17c
[INFO] onAprsisDataAvailable : # logresp BV5DJ-10 verified, server T2STRAS
[INFO] onAprsisDataAvailable : # aprsc 2.1.10-gd72a17c 22 Nov 2021 07:43:49 GMT T2STRAS 51.210.15.145:14580
[INFO] onAprsisDataAvailable : # aprsc 2.1.10-gd72a17c 22 Nov 2021 07:44:10 GMT T2STRAS 51.210.15.145:14580
[INFO] onAprsisDataAvailable : # aprsc 2.1.10-gd72a17c 22 Nov 2021 07:44:30 GMT T2STRAS 51.210.15.145:14580
[INFO] onAprsisDataAvailable : # aprsc 2.1.10-gd72a17c 22 Nov 2021 07:44:50 GMT T2STRAS 51.210.15.145:14580
[INFO] onAprsisDataAvai
```
[image]

sh123, 2021-11-22:
You cannot set power level to 22 dBm for this module, amplification must be set to no more than 30 (all gain) dBm - 12 dBm (amplifier) = 18 dBm, I was testing with 2dBm (+ 12 dBm amp) = +14 dBm.

GabeHC, 2021-11-22:
> You cannot set power level to 22 dBm for this module, it sets amplification for sx1268 controller only without amp, amplification must be set to no more than 30 (all gain) dBm - 12 dBm (amplifier) = 18 dBm, I was testing with 2dBm (+ 12 dBm amp) = +14 dBm.

the problem is  not receiving! I have no problem transmitting on power setting to 22 and measured power output at 1 watt

sh123, 2021-11-22:
Sorry, I cannot help, it works fine for me
{code}
10:13:16.735 -> [INFO] printConfig : Current mode: APRS-IS iGate
10:13:16.735 -> [INFO] printConfig : Built with RadioLib library
10:13:16.735 -> [INFO] printConfig : Using TNC KISS and AX.25 mode
10:13:16.735 -> [INFO] printConfig : UsbSerialEnable: no
10:13:16.735 -> [INFO] printConfig : EnableSignalReport: yes
10:13:16.735 -> [INFO] printConfig : EnablePersistentAprsConnection: yes
10:13:16.735 -> [INFO] printConfig : EnableRfToIs: yes
10:13:16.768 -> [INFO] printConfig : EnableIsToRf: no
10:13:16.768 -> [INFO] printConfig : EnableRepeater: no
10:13:16.768 -> [INFO] printConfig : EnableBeacon: no
10:13:16.768 -> [INFO] setupLora : Initializing LoRa
10:13:16.768 -> [INFO] setupLora : Frequency: 433750000 Hz
10:13:16.768 -> [INFO] setupLora : Bandwidth: 31250 Hz
10:13:16.768 -> [INFO] setupLora : Spreading: 10
10:13:16.768 -> [INFO] setupLora : Coding rate: 6
10:13:16.768 -> [INFO] setupLora : Power: 2 dBm
10:13:16.768 -> [INFO] setupLora : Sync: 0x34
10:13:16.801 -> [INFO] setupLora : CRC: enabled
10:13:16.801 -> [INFO] setupLora : Using SX126X module
10:13:16.801 -> [INFO] setupLora : LoRa initialized
10:13:16.834 -> [INFO] setup : Reading data on separate task
10:13:16.834 -> [INFO] setupWifi : WIFI connecting to WIFI_2G
10:13:16.834 -> [INFO] processIncomingDataTask : Incoming data process task started
10:13:17.532 -> [WARN] setupWifi : WIFI retrying 0
10:13:18.030 -> [WARN] setupWifi : WIFI retrying 1
10:13:18.527 -> [WARN] setupWifi : WIFI retrying 2
10:13:19.026 -> [WARN] setupWifi : WIFI retrying 3
10:13:19.525 -> [WARN] setupWifi : WIFI retrying 4
10:13:19.525 -> [INFO] setupWifi : WIFI connected to homenet
10:13:19.525 -> [INFO] setupWifi : IP address: 192.168.0.4
10:13:19.525 -> [INFO] setupBt : BT init loraprs_server
10:13:20.697 -> [INFO] setupBt : BT initialized
10:13:20.731 -> [INFO] reconnectAprsis : APRSIS connecting to rotate.aprs2.net
10:13:20.863 -> [INFO] reconnectAprsis : APRSIS connected
10:13:20.897 -> [INFO] reconnectAprsis : APRSIS logged in
10:13:21.395 -> [INFO] onAprsisDataAvailable : # aprsc 2.1.8-gf8824e8
10:13:21.894 -> [INFO] onAprsisDataAvailable : # logresp NOCALL-1 verified, server T2UKRAINE
10:13:41.432 -> [INFO] onAprsisDataAvailable : # aprsc 2.1.8-gf8824e8 22 Nov 2021 08:13:41 GMT T2UKRAINE 89.185.8.55:14580
10:14:01.531 -> [INFO] onAprsisDataAvailable : # aprsc 2.1.8-gf8824e8 22 Nov 2021 08:14:01 GMT T2UKRAINE 89.185.8.55:14580
10:14:21.627 -> [INFO] processIncomingRawPacketAsServer : NOCALL-7>APDR15,WIDE2-2:=/NN!!NN!!b sT 
10:14:21.627 -> [INFO] processIncomingRawPacketAsServer : Packet sent to APRS-IS
10:14:21.992 -> [INFO] onAprsisDataAvailable : # aprsc 2.1.8-gf8824e8 22 Nov 2021 08:14:20 GMT T2UKRAINE 89.185.8.55:14580
{code}

GabeHC, 2021-11-22:
that is incomplete copy and paste. it kept running but not hearing anything from TTGO tracker.

this is how I setup on "SX126x_Receive_Interrupt" and it works.
```
SX1268 radio = new Module(5, 33, 14, 39);
```   
and in setup
```
  radio.begin(438.2, 125.0, 12, 8, SX126X_SYNC_WORD_PRIVATE, 22, 32, 1.8, false);
  radio.setRegulatorDCDC(); 
  radio.autoLDRO();
  radio.setCurrentLimit(140);
  radio.setDio2AsRfSwitch(false);
  radio.explicitHeader();
  radio.setCRC(2);
  radio.setRfSwitchPins(2, 4);
  // set the function that will be called
  // when new packet is received
  radio.setDio1Action(setFlag);
  // start listening for LoRa packets0
  Serial.print(F("LORA Starting to listen ... "));
```
this is what I got from TTGO tracker
```
[SX1268] Received packet!
Got 1 Packages
<�BX4ACP-7>APLO02,WIDE1-1:!0000.00N/00000.00E[Lora Tracker Batt=4.09V
package length= 70/70 
[SX1268] RSSI:	0.00 dBm
[SX1268] SNR:	6.25 dB
```

Do you spot anything?

sh123, 2021-11-22:
```
  radio.setCRC(2);
```
They need to match between devices. Currently, I'm using setCRC(1).

sh123, 2021-11-22:
added config.h parameters for these values

GabeHC, 2021-11-22:
> added config.h parameters for these values

Great, it is working now, Sync word is the problem. 
Thank you so much for the help. 

it works for our tracker now, TTGO tracker has a non AX.25 payload warning. Do you know why?

[image]

sh123, 2021-11-22:
Good news! I'll change default sample value to 0x12 as it seems to be more correct. 0x34 is used for LoRaWAN networks.

This warning indicates that incoming packet could not be parsed. You can enable verbose logging then it will print incoming packets char by char, so can see raw contents.

GabeHC, 2021-11-22:
Polling mode works fine, for interrupt to work, ISR should only set flags, then process incoming data in the main loop. 

could you [email](bv5dj@hotmail.com) me your contact information? I would like to provide you one of our hardware for you to test with.

sh123, 2021-11-22:
Yes, when setting `CFG_LORA_USE_ISR` to `false` it will spawn separate ESP task for flag polling, which is set in ISR. Unfortunately, it is sub-optimal for battery powered applications and there is a risk of data loss if polling task blocks for longer period of time. Reading data in ISR works in RA-01 module, but not with E22 module. I talked to RadioLib maintainer about possibility to read and queue data in ISR, but he said that there is not guarantee for all boards and ESP32 flavors.

For me it is easier if you will post verbose packet output from the board, hardware delivery will take weeks and I have to go to customs.

sh123, 2021-11-22:
I'm closing it for now, as this problem seems to be solved, please, open another task for other issues.

closed 2021-11-22 by sh123