sh123/esp32_loraprs #29: Bluetooth Passthrough Option

karfam, 2021-11-11:
Would it be possible to add bluetooth passthrough ?

For example use a bluetooth terminal application and sent a text message between android phones.

sh123, 2021-11-11:
Do you mean to exclude KISS protocol? 

There is already such option `cfg.EnableTextPackets/CFG_TEXT_PACKETS`, https://github.com/sh123/esp32_loraprs/wiki/Protocol-and-compatibility

When enabled it will accept new line separated text messages without using KISS.

karfam, 2021-11-11:
Thanks for the quick reply Sir

I have read carefully your wiki and indeed use that option but thought it was only for use with aprsdroid. 

So maybe there is a slight change in the wiki like "Bluetooth Terminal Message Text : Enable this option" 

I can grab some screen shots if I may be of any help.

sh123, 2021-11-11:
Added note about Bluetooth terminal applications to wiki.

karfam, 2021-11-11:
Is there any debug logging?

I haven't made it work.

> 18:46:30.083 -> [INFO] printConfig : Current mode: NORMAL
18:46:30.083 -> [INFO] printConfig : Built with RadioLib library
18:46:30.083 -> [INFO] printConfig : Using TNC2 text mode
18:46:30.083 -> [INFO] printConfig : UsbSerialEnable: no
18:46:30.083 -> [INFO] setup : KISS extensions are disabled in TNC2 mode
18:46:30.083 -> [INFO] setupLora : Initializing LoRa
18:46:30.083 -> [INFO] setupLora : Frequency: 433775000 Hz
18:46:30.083 -> [INFO] setupLora : Bandwidth: 125000 Hz
18:46:30.083 -> [INFO] setupLora : Spreading: 12
18:46:30.083 -> [INFO] setupLora : Coding rate: 7
18:46:30.083 -> [INFO] setupLora : Power: 20 dBm
18:46:30.083 -> [INFO] setupLora : Sync: 0x34
18:46:30.083 -> [INFO] setupLora : CRC: enabled
18:46:30.083 -> [INFO] setupLora : LoRa initialized
18:46:30.083 -> [INFO] setupBt : BT init lora2
18:46:30.606 -> [INFO] setupBt : BT initialized 

I connect with the module through Bluetooth terminal App, but text message doesn't seem to pass to other module. Should I see something in the serial terminal when I pass a text message through Bluetooth terminal?

sh123, 2021-11-11:
There is no logging for data transfers, do you see that modem emits on sdr?

karfam, 2021-11-11:
I think there is no emit because it doesn't work even with aprsdroid kiss tnc. I will check with sdr shortly. 

I tried the modules tbeam with lily go github examples (with radiolib library) and transmit/receive is working.

ISR should be enabled? 

My modules are 1278.

sh123, 2021-11-11:
Strange, I tested this module with APRSDroid TCN2 mode and everything worked fine.

karfam, 2021-11-11:
> 20:36:56.831 -> ets Jul 29 2019 12:21:46
20:36:56.831 -> 
20:36:56.831 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:36:56.831 -> configsip: 0, SPIWP:0xee
20:36:56.831 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:36:56.878 -> mode:DIO, clock div:1
20:36:56.878 -> load:0x3fff0030,len:1100
20:36:56.878 -> ho 0 tail 12 room 4
20:36:56.878 -> load:0x40078000,len:12308
20:36:56.878 -> load:0x40080400,len:3076
20:36:56.878 -> entry 0x400805ec
20:36:57.390 -> [INFO] printConfig : Current mode: NORMAL
20:36:57.390 -> [INFO] printConfig : Built with RadioLib library
20:36:57.390 -> [INFO] printConfig : Using TNC KISS and AX.25 mode
20:36:57.390 -> [INFO] printConfig : UsbSerialEnable: no
20:36:57.390 -> [INFO] setupLora : Initializing LoRa
20:36:57.390 -> [INFO] setupLora : Frequency: 433775000 Hz
20:36:57.390 -> [INFO] setupLora : Bandwidth: 125000 Hz
20:36:57.390 -> [INFO] setupLora : Spreading: 12
20:36:57.390 -> [INFO] setupLora : Coding rate: 7
20:36:57.390 -> [INFO] setupLora : Power: 20 dBm
20:36:57.436 -> [INFO] setupLora : Sync: 0x34
20:36:57.436 -> [INFO] setupLora : CRC: enabled
20:36:57.436 -> [INFO] setupLora : LoRa initialized
20:36:57.436 -> [INFO] setupBt : BT init lora1
20:36:57.902 -> [INFO] setupBt : BT initialized

It transmits just fine with kiss tnc

[image]

It transmits fine with bluetooth terminal and text msg enabled

When  I use kiss tnc the red led  on the left side of the pcb blinks when a msg is send.
When I use text msg enable there is red led below gps module that blinks with 1hz. is this normal?

Also second module transmit just fine without notable drift.

BUT there is no decoding of the message from the other side. either way I send the message. I have the modules in different rooms.

karfam, 2021-11-11:
I have used your git in the past with the Lora library and the same modules worked just fined with kiss tnc and aprsdroid.

I am trying now to compile with Lora library (//#define USE_RADIOLIB) but it doesn't compile. 

The modules are ttgo t-beam 1.1v 1278 (433mhz)

sh123, 2021-11-12:
Not sure, try to use taskread branch with radiolib, which uses separate task to read data instead of interrupt based one. I never had problems with reading data. For loralib usage there was a bug just do git pull.

karfam, 2021-11-12:
According to official ttgo-tbeam IRQ pin seems to be 26? Or am I missing something?

Also you mean "this pin number should**NT** be defined anywhere else" ?

[image]

sh123, 2021-11-12:
Which module type do you select in Arduino IDE?

karfam, 2021-11-12:
> Which module type do you select in Arduino IDE?

[image]

sh123, 2021-11-12:
Are you sure it uses SX1278? Comments in pins_arduio.h are saying that it is SX1276.

karfam, 2021-11-12:
I am not sure if it uses sx1278.

I am using the following Ino from lily-go and receive - transmit functions work fine.

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/examples/RadioLibExamples/SX1278/SX1278_Receive

sh123, 2021-11-12:
Maybe try to set CFG_LORA_PIN_B to RADIO_BUSY_PIN, which is pin 32, so it will look like that
```
#define CFG_LORA_PIN_SS       SS // 18
#define CFG_LORA_PIN_RST      LORA_RST // 23
#define CFG_LORA_PIN_A        LORA_IRQ  // 26
#define CFG_LORA_PIN_A        32
```

karfam, 2021-11-12:
> #define CFG_LORA_PIN_SS       SS // 18
> #define CFG_LORA_PIN_RST      LORA_RST // 23
> #define CFG_LORA_PIN_A        LORA_IRQ  // 26
> #define CFG_LORA_PIN_B        32

Should I put the definitions in esp32_loraprs.ino ?

sh123, 2021-11-12:
All config parameters are set in config.h file, you can just replace RADIOLIB_NC with 32, I've also pushed some changes which will trace log incoming and outgoing data and config option to set log level, you need to set it to LVL_TRACE to see raw bytes transmitted and received.

karfam, 2021-11-12:
You are really helping very much in this problem and would like to thank you in advance.

**So to sum it up**

1. I should be using taskread branch for the tests
2. I define all the pins combinations in config.h
3. I set log level to LVL_TRACE for raw bytes

I will try all combinations/names so it may be help in the future for guys that have tbeam 1.1v and want to use radiolib

This is the pin definition according the seller.

[image]

sh123, 2021-11-12:
Added changes so you can use the same branch and set `CFG_LORA_USE_ISR` to `false` when using RadioLib, it will not read in ISR, but read on separate task in this case.

karfam, 2021-11-12:
What I have tried until now without success 

- (Sender) ESP32_loraaprs to (Receiver)SX1278_receive.ino --> decodes packet 
- Sender ESP32_loraaprs to ESP32_loraaprs --> no decoding ( ISR true and false)

The pins I use are this 

#define CFG_LORA_PIN_SS       18
#define CFG_LORA_PIN_RST      23
#define CFG_LORA_PIN_A        26    // (sx127x - dio0, sx126x/sx128x - dio1)
#define CFG_LORA_PIN_B        32 // (sx127x - dio1, sx126x/sx128x - busy)

LVL_TRACE should add up anything to serial when sending?

I recommend to open a different issue or change title of this one and add T-BEAM TTGO so maybe other users can report?

sh123, 2021-11-13:
Try to uncomment USE_RADIOLIB and disable ISR usage, I think it is just ISR, which is not triggered for some reason. This simple SX1278_receive example does not use anything except LoRa module, maybe HW has some conflicts when using both LoRa and Bluetooth.

sh123, 2021-11-13:
I think this pull request broke it https://github.com/sh123/esp32_loraprs/pull/28, I'll make a fix

karfam, 2021-11-13:
> I think this pull request broke it #28, I made a fix, please pull from master

You are right. latest master fixed the issue.
Decoding works fine along with Bluetooth passthrough.

Thanks for the support

// lora pinouts, CAD and ISR usage
#define CFG_LORA_PIN_SS       18
#define CFG_LORA_PIN_RST      23
#define CFG_LORA_PIN_A        26   // (sx127x - dio0, sx126x/sx128x - dio1)
#ifdef USE_RADIOLIB
#define CFG_LORA_PIN_B        32 // (sx127x - dio1, sx126x/sx128x - busy)
#endif
#define CFG_LORA_USE_ISR      true        // true - read incoming data in ISR, false - do not read in ISR
#define CFG_LORA_USE_CAD      false       // set to true to utilize carrier detection

sh123, 2021-11-13:
You are welcome.

closed 2021-11-13 by sh123