sh123/esp32_loraprs #30: Is it possible to create repeater modem?

happytm, 2021-12-12:
Is it possible to create a repeater node using ESP32 to relay messages between few clients to extend the range ? 

Ideally repeater modem consisting of ESP32 and LoRa chip will be mounted at higher location with powerful omnidirectional antenna and it's job will be to re transmit all the messages it receives via LoRa?

Thanks.

sh123, 2021-12-12:
If you mean APRS digirepeating then it is already supported, see you need to enable APRS iGate mode and set `CFG_DIGIREPEAT` to `true`, https://github.com/sh123/esp32_loraprs/wiki/Configuration-parameters

happytm, 2021-12-12:
Thank you for your quick reply. I was basically requesting repeater for DV traffic for walkie talkies using you android app and ESP32-loraprs modems. It will be beneficial even if I can use 3 ESP32 modems - 2 connected to android phones and 3rd as a repeater so range could be at least doubled.

Thanks.

sh123, 2021-12-12:
To operate repeater need two modems and duplexer if operating on the same band, one modem is used for RX and another for simultaneous TX. Otherwise, modem need to operate as parrot, recording what's received and then transmit on the same channel.

happytm, 2021-12-12:
Understood. Thanks.

sh123, 2021-12-12:
Implementing such repeater is very easy, you just need to read raw data from one modem on RX frequency and transmit to another one on TX frequency in a loop. It will take some 10 lines of code. But, client modems or applications should support split operation, they need to listen on one frequency, but transmit on another. Also, you will need some kind of duplexer if using the same antenna for transmit and receive or use two separate RX and TX antennas and move them further apart from each other. 

Implementing simplex repeater (parrot) is also easy, just read data from the modem and record it to the in memory buffer, when no data is arrived during some time, then just retransmit it back on the same frequency.

happytm, 2021-12-12:
Thanks for detailed answer.

closed 2022-02-12 by sh123