sh123/esp32_loraprs #32: Pinout

madbird99, 2022-02-24:
Hello, would you be so kind as to tell me the connections between the pins of a wroom-32 38 pins and an rfm95?
Thanks in advance.

sh123, 2022-02-25:
[image]

madbird99, 2022-02-26:
Thanks for your answer but I was wrong, my lora module is rfm98, based on sx1278. The unmodified config.h corresponds to this pinout for my module? 

    Common SPI:
        MOSI: GPIO_23/VSPI_MOSI
        MISO: GPIO_19/VSPI_MISO
        SCK: GPIO_18/VSPI_SCK
 
    Board specific:
        SS/CS/NSS: GPIO_5
        RST/RESET: GPIO_26
        DIO0/IRQ: GPIO_14

sh123, 2022-02-26:
Yes, these are default values if you are using WROOM ESP32 board, for other boards it tries to use board specific values taken from `pins_arduino.h` file, which is part of Arduino IDE tools and which is used when you select ESP32 board type in IDE.
```
#ifndef LORA_RST
#pragma message("LoRa pin definitions are not found, redefining...")
#define LORA_RST              26
#define LORA_IRQ              14
#endif

// LoRa pinouts
#define CFG_LORA_PIN_SS       SS
#define CFG_LORA_PIN_RST      LORA_RST
#define CFG_LORA_PIN_A        LORA_IRQ    // (sx127x - dio0, sx126x/sx128x - dio1)
```