sh123/esp32_loraprs #33: iOS software "APRS.FI" cannot work

pepefrog1234, 2022-06-16:
This is my setting。
`#define CFG_BT_USE_BLE        true `
My APRS.FI can be connected to ESP32 (BLE), but the app sends package, ESP32 does not work (TX LORA signal)
[image]

```
#define LED_TOGGLE_PERIOD     1000

#define SERIAL_BAUD_RATE      115200

// change pinouts if not defined through native board LORA_* definitions
#define USE_RADIOLIB

#ifndef LORA_RST
#pragma message("LoRa pin definitions are not found, redefining...")
#define LORA_RST              14
#define LORA_IRQ              33
#endif

#ifndef BUILTIN_LED
#pragma message("BUILDIN_LED is not found, defining as 12")
#define BUILTIN_LED           12
#endif
//#define BUILTIN_LED           12
#define CFG_IS_CLIENT_MODE    true        // false - server mode (APRS-IS gate mode)

// lora pinouts, CAD and ISR usage
#define CFG_LORA_PIN_SS       SS
#define CFG_LORA_PIN_RST      LORA_RST
#define CFG_LORA_PIN_DIO0     LORA_IRQ
#ifdef USE_RADIOLIB
#define CFG_LORA_PIN_DIO1     LORA_IRQ    // set to your DIO1 pin number if connected
#define CFG_LORA_USE_ISR      true        // always ON for RadioLib
#else
#define CFG_LORA_PIN_DIO1     RADIOLIB_NC   // not used in arduino-LoRa
#define CFG_LORA_USE_ISR      false       // set to true for ISR usage in arduino-LoRa
#endif
#define CFG_LORA_USE_CAD      true        // set to true to utilize carrier detection

// lora protocol parameters
#define CFG_LORA_FREQ         432.19E6 //改 430.64
#define CFG_LORA_BW           125e3
#define CFG_LORA_SF           12
#define CFG_LORA_CR           8
#define CFG_LORA_PWR          30
#define CFG_LORA_ENABLE_CRC   true        // set to false for speech data

// wifi client and AP options
#define CFG_WIFI_ENABLE_AP    false       // run as wifi access point, for CFG_KISS_TCP_IP mode
#define CFG_WIFI_SSID         "PTY"      // connect to SSID or run as this SSID in AP mode
#define CFG_WIFI_KEY          "0422412908"  // wifi key

// bluetooth
#define CFG_BT_NAME           "loraprs"
#define CFG_BT_USE_BLE        true        // set to true to use bluetooth low energy (for ios devices)

// USB
#define CFG_USB_SERIAL_ENABLE false       // true - enable communication over USB Serial

// KISS protocol options
#define CFG_KISS_EXTENSIONS   true    // true - enable modem control from application with KISS commands 改ture
#define CFG_KISS_TCP_IP       true   // true - run as KISS TCP/IP server, no bluetooth operations performed

// APRS-IS options, valid in when CFG_IS_CLIENT_MODE = false
#define CFG_APRS_LOGIN        "BX4ACP-10"
#define CFG_APRS_PASS         "18091"
#define CFG_APRS_FILTER	      "r/24.11/120.42/50"
//#define CFG_APRS_RAW_BKN      "BV5DJ-10>APZMDM,WIDE1-1:!0000.00N/00000.00E#LoRA 433.775MHz/BW125/SF12/CR7/0x34"
#define CFG_APRS_RAW_BKN      "BX4ACP>APZMDM,WIDE1-1:!2410.70N/12043.00E#LoRA 438.2MHz/BW125/SF12/CR8/0x12"
// APRS-IS gateway options, valid in when CFG_IS_CLIENT_MODE = false
#define CFG_PERSISTENT_APRS   false   // keep tcp/ip connection open (lot of traffic), otherwise connect on new packet (very rare traffic)
#define CFG_DIGIREPEAT        false   // digirepeat incoming packets
#define CFG_RF_TO_IS          true    // forward packets from radio to internet
#define CFG_IS_TO_RF          false    // forward packets from internet to radio basedon CFG_APRS_FILTER
#define CFG_BEACON            false    // enable perdiodc beacon from CFG_APRS_RAW_BKN

// frequency correction for narrow band bandwidths
#define CFG_FREQ_CORR         false   // true - correct own frequency based on received packet
#define CFG_FREQ_CORR_DELTA   1000    // correct when frequency difference is larger than this value

// PTT control
#define CFG_PTT_ENABLE        false   // enable external ptt (relay) control (for amplifier)
#define CFG_PTT_PIN           4       // PTT pin
#define CFG_PTT_TX_DELAY_MS   50      // time between relay swited on and transmission
#define CFG_PTT_TX_TAIL_MS    10      // time between stop transmission and relay off`
```

sh123, 2022-06-16:
@pepefrog1234 , maybe because you have
`
#define CFG_KISS_TCP_IP       true   // true - run as KISS TCP/IP server, no bluetooth operations performed
`

pepefrog1234, 2022-06-16:
I just tried to set the `#define CFG_KISS_TCP_IP to false`, but I still don’t work.

sh123, 2022-06-17:
@pepefrog1234 , you need to provide more information.
- Which board are you using? Are pinout definition correct?
- Does it work with APRSDroid?
- Do you see transmission on the spectrum analyzer?

pepefrog1234, 2022-06-17:
1.I'm using the ESP32 pins to make sure it's correct, as I'm using traditional bluetooth serial transmission which is normal.
2.APRSDroid serial bluetooth can work normally when BLE function is turned off
3.NO.
This is my config using the latest version
```
// comment out for arduino-Lora usage
#define USE_RADIOLIB

// Uncomment for SX126X module usage
#define USE_SX126X

// Check your module name at https://github.com/jgromes/RadioLib/wiki/Modules
#ifdef USE_SX126X
#define MODULE_NAME   SX1268
#else
#define MODULE_NAME   SX1278
#endif

// generic options
#define LED_TOGGLE_PERIOD     1000    // heartbeat LED period
#define SERIAL_BAUD_RATE      115200  // USB serial baud rate

// USB serial logging
// set to DebugLogLevel::LVL_TRACE for packet logging
// set to DebugLogLevel::LVL_NONE to disable logging
#define CFG_LOG_LEVEL         DebugLogLevel::LVL_INFO

// select between client mode and APRS-IS gate mode (客戶端)
#define CFG_IS_CLIENT_MODE    true        // false - server mode (APRS-IS gate mode)

// change pinouts if not defined through native board LORA_* definitions
#ifndef LORA_RST
#pragma message("LoRa pin definitions are not found, redefining...")
#define LORA_RST              14
#define LORA_IRQ              33
#endif

// LoRa pinouts
#define CFG_LORA_PIN_SS       SS
#define CFG_LORA_PIN_RST      LORA_RST
#define CFG_LORA_PIN_A        12    // (sx127x - dio0, sx126x/sx128x - dio1)
#ifdef USE_SX126X
#define CFG_LORA_PIN_B        39          // (sx127x - dio1, sx126x/sx128x - busy)
#define CFG_LORA_PIN_RXEN     2           // (sx127x - unused, sx126x - RXEN pin number)
#define CFG_LORA_PIN_TXEN     4           // (sx127x - unused, sx126x - TXEN pin number)
#else
#define CFG_LORA_PIN_B        16 // (sx127x - dio1, sx126x/sx128x - busy)
#endif

// Redefine LED if not defined in Arduino to have module heartbeat indication
#ifndef BUILTIN_LED
#pragma message("BUILDIN_LED is not found, defining as 2")
#define BUILTIN_LED           12
#endif

// CAD and ISR usage selection
#ifdef USE_SX126X
#define CFG_LORA_USE_ISR      false       // reading in ISR does not work on sx126x
#define CFG_LORA_USE_CAD      true        // do not transmit if channel is busy
#else
#define CFG_LORA_USE_ISR      true        // true - read incoming data in ISR, false - do not read in ISR
#define CFG_LORA_USE_CAD      true        // set to true to utilize carrier detection
#endif

// LoRa protocol default parameters (they need to match between devices!!!)
#define CFG_LORA_FREQ_RX      432.19e6   // RX frequency in MHz
#define CFG_LORA_FREQ_TX      432.19e6   // TX frequency in MHz
#define CFG_LORA_BW           125e3       // bandwidth (from 7.8 kHz up to 500 kHz)
#define CFG_LORA_SF           12          // spreading factor (6 - 12), 6 requires implicit header mode
#define CFG_LORA_CR           8           // coding rate (5 - 8)
[... 60 lines cut ...]
[INFO] printConfig : Current mode: NORMAL
[INFO] printConfig : Built with RadioLib library
[INFO] printConfig : Using TNC KISS and AX.25 mode
[INFO] printConfig : UsbSerialEnable: no
[INFO] setupLora : Initializing LoRa
[INFO] setupLora : Frequency: 432190000 Hz
[INFO] setupLora : Bandwidth: 125000 Hz
[INFO] setupLora : Spreading: 12
[INFO] setupLora : Coding rate: 8
[INFO] setupLora : Power: 22 dBm
[INFO] setupLora : Sync: 0x34
[INFO] setupLora : CRC: 2
[INFO] setupLora : Header: explicit
[INFO] setupLora : Speed: 183 bps
[INFO] setupLora : Min level: -137.03 dBm
[INFO] setupLora : Using SX126X module
[INFO] setupLora : LoRa initialized
[INFO] setup : Reading data on separate task
[INFO] setupBt : BLE init 613
[INFO] processIncomingDataTask : Incoming data process task started
[INFO] begin : starting service
[INFO] begin : Waiting a client connection to notify...
[INFO] setupBt : BLE initialized
```
[pcb configuration](https://oshwlab.com/GabeH/esp32-lora-aprs)

sh123, 2022-06-17:
I never tested BLE myself, maybe @mrpatrick1991 can help as he was adding BLE support. I'll try to find iphone to test a bit later if he is offline.

sh123, 2022-06-18:
@pepefrog1234 , I tested BLE with Android NRF application (https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) and I was able to transmit KISS ax25 byte array over BLE, I set log level to trace as
`
#define CFG_LOG_LEVEL         DebugLogLevel::LVL_TRACE
`
and from the log I could see bytes received from BLE and transmitted to the radio module, also observed transmission on the spectrogram
`
12:22:58.439 -> [TRACE] onSerialRx : ⸮ c0
12:22:58.439 -> [TRACE] onSerialRx : 0
12:22:58.439 -> [TRACE] onSerialRx : ⸮ 82
12:22:58.439 -> [TRACE] onRigTx : ⸮ 82
12:22:58.439 -> [TRACE] onSerialRx : ⸮ a0
12:22:58.439 -> [TRACE] onRigTx : ⸮ a0
`
I do not have iPhone near by at the moment, so maybe there is something to do with how iPhone handles BLE.

pepefrog1234, 2022-06-18:
I've also tried receiving packets with ble, but it still doesn't work. You may really need to use ios for testing. Because I have tried all kinds of methods here, the software just doesn't work (including the PocketPacket alternative).

sh123, 2022-06-18:
Do you see anything received from bluetooth when LVL_TRACE is enabled?

pepefrog1234, 2022-06-18:
Absolutely not.
I use sdr to observe the spectrum, and I can't find any signal.

sh123, 2022-06-18:
Try master version, I've added
```
    pServer->getAdvertising()->setScanResponse(true);
    pServer->getAdvertising()->setMinPreferred(0x06);
    pServer->getAdvertising()->setMaxPreferred(0x12);
```
into `ble_serial.cpp` BLE initialization code, which is recommended for iOS.

pepefrog1234, 2022-06-18:
Still can't work, this is the log file.
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[INFO] printConfig : Current mode: NORMAL
[INFO] printConfig : Built with RadioLib library
[INFO] printConfig : Using TNC KISS and AX.25 mode
[INFO] printConfig : UsbSerialEnable: no
[INFO] setupLora : Initializing LoRa
[INFO] setupLora : Frequency: 432190000 Hz
[INFO] setupLora : Bandwidth: 125000 Hz
[INFO] setupLora : Spreading: 12
[INFO] setupLora : Coding rate: 8
[INFO] setupLora : Power: 22 dBm
[INFO] setupLora : Sync: 0x34
[INFO] setupLora : CRC: 2
[INFO] setupLora : Header: explicit
[INFO] setupLora : Speed: 183 bps
[INFO] setupLora : Min level: -137.03 dBm
[INFO] setupLora : Using SX126X module
[INFO] setupLora : LoRa initialized
[INFO] setup : Reading data on separate task
[INFO] setupBt : BLE init 613
[INFO] processIncomingDataTask : Incoming data process task started
[INFO] begin : starting service
[INFO] begin : Waiting a client connection to notify...
[INFO] setupBt : BLE initialized
[INFO] onDisconnect : Started advertising
[INFO] onDisconnect : Started advertising
[INFO] onDisconnect : Started advertising
[INFO] onDisconnect : Started advertising
```

pepefrog1234, 2022-06-19:
@sh123 To day I just tested and can receive packets but still can't send packets from the phone

sh123, 2022-06-19:
If you see 
```
INFO] onDisconnect : Started advertising
```
then it means that client is able to connect, this line is sent when client disconnects, if TRACE logging is enabled and it does not show anything received then there must be some problem with the app not sending data, are you able to get it working with some other modem? Because I can transmit using nRF Android utility.

pepefrog1234, 2022-06-20:
There is a line of INFO] onDisconnect : Started advertising, the application shows that the connection with tnc is normal, but it cannot work. Even if I use other apps from the app store.

pepefrog1234, 2022-06-20:
I just tried it, send [ByteArray] in [DebugLogLevel::LVL_TRACE] mode, the serial window of the Bluetooth modem has a printed string
[image]
Also I found that the bluetooth modem can receive the beacons transmitted by my igate station.
[image]

sh123, 2022-06-20:
If it works with nRF application, but does not work with aprs.fi then send TRACE output from the console when trying to transmit with aprs.fi application to check KISS/ax25 packet format.

sh123, 2022-06-20:
There is no source code for aprs.fi application, so cannot say for sure, maybe it tries to use different characteristic, currently modem is using `00000001-ba2a-46c9-ae49-01b0961f68bb` service UUID and `00000002-ba2a-46c9-ae49-01b0961f68bb` WRITE characteristic for phone -> modem transmission as written in specification.

sh123, 2022-06-20:
I don't know.

pepefrog1234, 2022-06-20:
OK, I understand, thank you for your assistance

closed 2022-08-25 by sh123