sh123/esp32_loraprs #34: Broken txt aprs packet?

tortyl, 2022-08-24:
Hello

I was try use igate to recevice txt aprs from Lora Tracker from this project: https://github.com/lora-aprs/LoRa_APRS_iGate.
My setup: Lora RA-02 trx + TTGO T7 v1.3.
I was know to change config: 
`#define CFG_TEXT_PACKETS      true`

Broken(?) packet received over LORA from tracker:
```
[TRACE] onSerialTx : < 3c
[TRACE] onSerialTx : ⸮ ff
[TRACE] onSerialTx :  1
[TRACE] onSerialTx : S 53
[TRACE] onSerialTx : Q 51
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : N 4e
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx : 7 37
[TRACE] onSerialTx : > 3e
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx : P 50
[TRACE] onSerialTx : L 4c
[TRACE] onSerialTx : T 54
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : , 2c
[TRACE] onSerialTx : W 57
[TRACE] onSerialTx : I 49
[TRACE] onSerialTx : D 44
[TRACE] onSerialTx : E 45
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : 5 35
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 8 38
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 2 32
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : N 4e
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 7 37
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 8 38
[TRACE] onSerialTx : E 45
[TRACE] onSerialTx : [ 5b
[TRACE] onSerialTx : 3 33
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 3 33
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx : = 3d
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx :   20
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : _ 5f
[TRACE] onSerialTx : B 42
[TRACE] onSerialTx : a 61
[TRACE] onSerialTx : t 74
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : 3 33
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : V 56
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : u 75
[TRACE] onSerialTx : r 72
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : m 6d
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : w 77
[TRACE] onSerialTx : 8 38
[TRACE] onSerialTx : R 52
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : 
 a
```

This packet is not valid and is not transmited to APRS-IS
I was try to debug and seen that source and destination calls was not recognized:
`Serial.println(payload.ToString());`

```
>:!5108.21N/01647.48E[360/003/A=000464 -  _Bat.: 3.66V - Cur.: 644mA !w8R
[WARN] processIncomingRawPacketAsServer : Skipping non-AX25 payload
```

What can i do?

Best regards
Tom

sh123, 2022-08-25:
Packet starts with:
```
[TRACE] onSerialTx : < 3c
[TRACE] onSerialTx : ⸮ ff
[TRACE] onSerialTx :  1
```
so it just fails to parse call sign, because of this data, which makes call sign invalid.

Looks like Lora_APRS_iGate prefix their packets with some identified:
```
// cppcheck-suppress unusedFunction
void LoRa_APRS::sendMessage(const std::shared_ptr<APRSMessage> msg)
{
	setFrequency(_TxFrequency);
	String data = msg->encode();
	beginPacket();
	// Header:
	write('<');
	write(0xFF);
	write(0x01);
	// APRS Data:
	write((const uint8_t *)data.c_str(), data.length());
	endPacket();
	setFrequency(_RxFrequency);
}
```

I'll update code to teke it into account.

sh123, 2022-08-25:
@tortyl , pull from master, added change.

closed 2022-08-25 by sh123

tortyl, 2022-08-25:
Unfortunately it still doesn't work :(
The improvement is visible, but the problem is still there.

```
[TRACE] onSerialTx : S 53
[TRACE] onSerialTx : Q 51
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : N 4e
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx : 7 37
[TRACE] onSerialTx : > 3e
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx : P 50
[TRACE] onSerialTx : L 4c
[TRACE] onSerialTx : T 54
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : , 2c
[TRACE] onSerialTx : W 57
[TRACE] onSerialTx : I 49
[TRACE] onSerialTx : D 44
[TRACE] onSerialTx : E 45
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : 5 35
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 8 38
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 2 32
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : N 4e
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 7 37
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 5 35
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : E 45
[TRACE] onSerialTx : [ 5b
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 3 33
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx : = 3d
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : L 4c
[TRACE] onSerialTx : o 6f
[TRACE] onSerialTx : R 52
[TRACE] onSerialTx : a 61
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : T 54
[TRACE] onSerialTx : r 72
[TRACE] onSerialTx : a 61
[TRACE] onSerialTx : c 63
[TRACE] onSerialTx : k 6b
[TRACE] onSerialTx : e 65
[TRACE] onSerialTx : r 72
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx :   20
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : _ 5f
[TRACE] onSerialTx : B 42
[TRACE] onSerialTx : a 61
[TRACE] onSerialTx : t 74
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 5 35
[TRACE] onSerialTx : V 56
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : u 75
[TRACE] onSerialTx : r 72
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx :   20
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : m 6d
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : w 77
[TRACE] onSerialTx : u 75
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : 
 a
>:!5108.20N/01647.50E[134/000/A=000466LoRa Tracker -  _Bat.: 4.15V - Cur.:  0mA !wu:
[WARN] processIncomingRawPacketAsServer : Skipping non-AX25 payload
```

sh123, 2022-08-25:
added some changes, try

tortyl, 2022-08-26:
Now everything works :)

```
[TRACE] onSerialTx : S 53
[TRACE] onSerialTx : Q 51
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : N 4e
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx : 7 37
[TRACE] onSerialTx : > 3e
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx : P 50
[TRACE] onSerialTx : L 4c
[TRACE] onSerialTx : T 54
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : , 2c
[TRACE] onSerialTx : W 57
[TRACE] onSerialTx : I 49
[TRACE] onSerialTx : D 44
[TRACE] onSerialTx : E 45
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : 5 35
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 8 38
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 2 32
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : N 4e
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 7 37
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 9 39
[TRACE] onSerialTx : E 45
[TRACE] onSerialTx : [ 5b
[TRACE] onSerialTx : 3 33
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : / 2f
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx : = 3d
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 0 30
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : 7 37
[TRACE] onSerialTx : 6 36
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx :   20
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : _ 5f
[TRACE] onSerialTx : B 42
[TRACE] onSerialTx : a 61
[TRACE] onSerialTx : t 74
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : 4 34
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : 1 31
[TRACE] onSerialTx : V 56
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : C 43
[TRACE] onSerialTx : u 75
[TRACE] onSerialTx : r 72
[TRACE] onSerialTx : . 2e
[TRACE] onSerialTx : : 3a
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : - 2d
[TRACE] onSerialTx : 8 38
[TRACE] onSerialTx : 9 39
[TRACE] onSerialTx : m 6d
[TRACE] onSerialTx : A 41
[TRACE] onSerialTx :   20
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : w 77
[TRACE] onSerialTx : [ 5b
[TRACE] onSerialTx : i 69
[TRACE] onSerialTx : ! 21
[TRACE] onSerialTx : 
 a
SQ6NCC-7>APLT00,WIDE1-1:!5108.21N/01647.49E[360/000/A=000476 -  _Bat.: 4.11V - Cur.: -89mA !w[i
[INFO] processIncomingRawPacketAsServer : SQ6NCC-7>APLT00,WIDE1-1:!5108.21N/01647.49E[360/000/A=000476 -  _Bat.: 4.11V - Cur.: -89mA !w[irssi: -60.00dBm, snr: 7.75dB, err: -2672Hz
[INFO] processIncomingRawPacketAsServer : Packet sent to APRS-IS
```

Could you check if the beacon via LORA is transmitting correctly?
I caught such a thing in the debug:

```
[INFO] sendPeriodicBeacon : Periodic beacon is sent
[TRACE] onRigTx : < 3c
[TRACE] onRigTx : ⸮ ff
[TRACE] onRigTx :  1
[TRACE] onRigTx : S 53
[TRACE] onRigTx : Q 51
[TRACE] onRigTx : 6 36
[TRACE] onRigTx : N 4e
[TRACE] onRigTx : C 43
[TRACE] onRigTx : C 43
[TRACE] onRigTx : - 2d
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 0 30
[TRACE] onRigTx : > 3e
[TRACE] onRigTx : A 41
[TRACE] onRigTx : P 50
[TRACE] onRigTx : Z 5a
[TRACE] onRigTx : M 4d
[TRACE] onRigTx : D 44
[TRACE] onRigTx : M 4d
[TRACE] onRigTx : , 2c
[TRACE] onRigTx : W 57
[TRACE] onRigTx : I 49
[TRACE] onRigTx : D 44
[TRACE] onRigTx : E 45
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : - 2d
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : : 3a
[TRACE] onRigTx : ! 21
[TRACE] onRigTx : 5 35
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 0 30
[TRACE] onRigTx : 8 38
[TRACE] onRigTx : . 2e
[TRACE] onRigTx : 2 32
[TRACE] onRigTx : 2 32
[TRACE] onRigTx : N 4e
[TRACE] onRigTx : L 4c
[TRACE] onRigTx : 0 30
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 6 36
[TRACE] onRigTx : 4 34
[TRACE] onRigTx : 7 37
[TRACE] onRigTx : . 2e
[TRACE] onRigTx : 4 34
[TRACE] onRigTx : 9 39
[TRACE] onRigTx : E 45
[TRACE] onRigTx : & 26
[TRACE] onRigTx : L 4c
[TRACE] onRigTx : o 6f
[TRACE] onRigTx : R 52
[TRACE] onRigTx : A 41
[TRACE] onRigTx :   20
[TRACE] onRigTx : 4 34
[TRACE] onRigTx : 3 33
[TRACE] onRigTx : 3 33
[TRACE] onRigTx : . 2e
[TRACE] onRigTx : 7 37
[TRACE] onRigTx : 7 37
[TRACE] onRigTx : 5 35
[TRACE] onRigTx : M 4d
[TRACE] onRigTx : H 48
[TRACE] onRigTx : z 7a
[TRACE] onRigTx : / 2f
[TRACE] onRigTx : B 42
[TRACE] onRigTx : W 57
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 2 32
[TRACE] onRigTx : 5 35
[TRACE] onRigTx : / 2f
[TRACE] onRigTx : S 53
[TRACE] onRigTx : F 46
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 2 32
[TRACE] onRigTx : / 2f
[TRACE] onRigTx : C 43
[TRACE] onRigTx : R 52
[TRACE] onRigTx : 5 35
[TRACE] onRigTx : / 2f
[TRACE] onRigTx : 0 30
[TRACE] onRigTx : x 78
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 
```

I don't have a second device to confirm if the packet is correct.

sh123, 2022-08-26:
Yes, I've added this prefix for transmission to make it compatible with lora-aprs.info trackers, so they can receive your packet.