sh123/esp32_loraprs #39: Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)

DerHirschi, 2022-10-16:
I've got some crashes like u see in the topic.

I can't verifying crashes on your original version because i had to modify the sketch a bit so that it works with RadioLib and SX1276 ( 868 MHz ).

But it seems the crashes decreasing when i increase CfgPollDelayMs to 10. Now i have set it to 15 and it seems ok but i still testing.. ( Get still some losts of ip connection from the Modul but i can't tell if it's related to the crash or just to a weak wifi connection )

Another think i did, but don't laugh, i am just a amateur, i commented out 
delay(CfgPollDelayMs);
in
Service::processIncomingDataTask
.

I came to the decision to do so,  because i googled the problem.
It seems to came to this problem when u delay a secondary task ( like a thread, like i understand).
And like i understand Service::processIncomingDataTask is a secondary task in ISR Mode.

Like i said.. This both changes seemed to work on my side. I also discovered that the Moduls are really hot after they crashed..

So it makes sens to increase CfgPollDelayMs to get load from the CPU and it wouldn't be necessary to comment out the  delay(CfgPollDelayMs) in Service::processIncomingDataTask .

sh123, 2022-10-16:
Radio lib does not allow reading data in ISR, so it sets boolean flag in ISR and uses another task for data reading. Currently, it is just polling for this variable and uses delay to decrease CPU cycles wasted. Ideally, it needs to use some esp32 primitives to be notified without polling, such as events.

DerHirschi, 2022-10-16:
Ok, 
i keep an eye on this "crash issues" but to increase loop delay seems working.
For me the Issues is closed for now.

Thanks..

sh123, 2022-10-16:
I updated code to use xTaskNotifyFromISR/xTaskNotifyWait pair and removed polling in RX task. It is more optimal and now polling interval in the main loop can me increased without risk of RX data loss.

closed 2022-10-16 by sh123