sh123/esp32_loraprs #41: Packets not reach APRS-IS network

Nautiluscl, 2022-12-01:
Hello

Thank you for your hard work coding this project. I succesfully flashed a TTGO Lora32, to use it by a standalone iGate, its working OK except for the APRS-IS connection. The packets dont reach the APRS-IS network, looks like they are being dropped by the APRSC Server. 

I tried with the server of a friend ( http://radioaficion.pro:14501/ ) and noticed the packet RX count sent by the iGate on red on the client list. 

I dont know what is wrong. Any help is apreciated

Alejandro CE3VRT

sh123, 2022-12-01:
@Nautiluscl , what is inside the USB console log? Does it work with the default aprs gate?

Nautiluscl, 2022-12-01:
Hello, thanks for your quick reply.

1.-Since im using a board with SX127x, i commented this:

// Uncomment for SX126X module usage
//#define USE_SX126X

And got a compilation error: 'CFG_LORA_PIN_RXEN' was not declared in this scope . I managed to clear it moving this:

```
#define CFG_LORA_PIN_RXEN     4           // (sx127x - unused, sx126x - RXEN pin number)
#define CFG_LORA_PIN_TXEN     5           // (sx127x - unused, sx126x - TXEN pin number)
```

outside the #ifdef USE_SX126X . Then it compiled fine  (dont know if it the best solution but worked for me)

My intention is to use this project as a standalone iGate, on a TTGO Lora32-OLED V2.1.6 board. For the tracker im using a TTGO T-beam tracker loaded with the project from https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS and is working fine.

So i set this:

```
// select between client mode and APRS-IS gate mode
#define CFG_IS_CLIENT_MODE    false        // false - server mode (APRS-IS gate mode)
```

Then i configured the APRS-IS options:

```
// APRS-IS options, valid in when CFG_IS_CLIENT_MODE = false
#define CFG_APRS_LOGIN        "CE3VRT-1"
#define CFG_APRS_PASS         "20901"
#define CFG_APRS_FILTER	      "r/35.60/139.80/25"  // multiple are space separated, see http://www.aprs-is.net/javAPRSFilter.aspx
#define CFG_APRS_RAW_BKN      "CE3VRT-1>APZMDM,WIDE1-1:!3336.89S/07042.53W#LoRA 433.775MHz/BW125/SF12/CR7/0x12"

// APRS-IS gateway options, valid when CFG_IS_CLIENT_MODE = false
#define CFG_SIGNAL_REPORT     true    // include signal report into the comment when sending to APRS-IS
#define CFG_PERSISTENT_APRS   true    // keep tcp/ip connection open (lot of traffic), otherwise connect on new packet (very rare traffic)
#define CFG_DIGIREPEAT        false   // digirepeat incoming packets
#define CFG_RF_TO_IS          true    // forward packets from radio to internet
#define CFG_IS_TO_RF          false   // forward packets from internet to radio based on CFG_APRS_FILTER
#define CFG_BEACON            true   // enable perdiodic beacon from CFG_APRS_RAW_BKN
#define CFG_TEXT_PACKETS      true   // enable aprs text packets instead of binary for interoperability with other projects (disables KISS + AX.25!)

```
This is shown on the USB serial console: 

```
[INFO] setupWifi : IP address: 192.168.100.190
[INFO] reconnectAprsis : APRSIS connecting to rotate.aprs2.net
[INFO] reconnectAprsis : APRSIS connected
[INFO] reconnectAprsis : APRSIS logged in
[INFO] sendPeriodicBeacon : Periodic beacon is sent
[INFO] onAprsisDataAvailable : # aprsc 2.1.4-g408ed49
[INFO] onAprsisDataAvailable : # logresp CE3VRT-1 verified, server T2CSNGRAD
[INFO] onAprsisDataAvailable : # aprsc 2.1.4-g408ed49 1 Dec 2022 16:34:09 GMT T2CSNGRAD 85.90.180.26:14580
[INFO] processIncomingRawPacketAsServer : CE3VRT-6>APLO02*:!/_M>3<Y_=S&"HT-Beam tracker Batt=3.91Vrssi: -79.00dBm, snr: 10.50dB, err: 2324Hz
[INFO] processIncomingRawPacketAsServer : Packet sent to APRS-IS
```

CE3VRT-6 is my tracker, so it seems like the RF side works ok! it receive the packets, but they dont reach the APRS-IS network, i dont see them on aprs.fi or PinPoint APRS software. The perodic beacon also dont reach the network, i dont see it on the map, but the beacon TX from the board fine, i see is RX'ed by the tracker fine.

Thank you for your help

sh123, 2022-12-01:
I see some packets on https://aprs.fi/?c=raw&call=CE3VRT-6, are they from this tracker?

Nautiluscl, 2022-12-01:
> I see some packets on https://aprs.fi/?c=raw&call=CE3VRT-6, are they from this tracker?

Yes, but i was using another iGate... I was testing the firmware from [SQ9MDD](https://github.com/SQ9MDD) only as KISS TNC and using PinPoint APRS as igate. Also tested APRX on a raspberry sucessfully.

sh123, 2022-12-01:
Maybe it does not like long comment, try to disable signal report.

sh123, 2022-12-01:
Added small fix, try. Also, digirpeating most likely is not gonna work for E-Mic packets with this project if path is encoded inside dist callsign part.

Nautiluscl, 2022-12-01:
> Added small fix, try. Also, digirpeating most likely is not gonna work for E-Mic packets by this project if path is encoded inside destination callsign part, it uses generic digi path only.

It worked! (i changed the callsign to CE3VRT-2 to avoid confusions) 

[image]

Thank you!

closed 2022-12-01 by sh123