sh123/esp32_loraprs #44: 'Radio start error: -13' and 'GPIO isr service is not installed, call gpio_install_isr_service() first'

yooshky, 2022-12-09:
What is wrong? Errors:
1. setupRig : Radio start error: -13
2. gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first

From console:
```rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[INFO] printConfig : Current mode: NORMAL
[INFO] printConfig : Using TNC KISS and AX.25 mode
[INFO] printConfig : UsbSerialEnable: no
[INFO] setupRig : Initializing LoRa
[INFO] setupRig : Frequency: 434855000 Hz
[INFO] setupRig : Bandwidth: 125000 Hz
[INFO] setupRig : Spreading: 9
[INFO] setupRig : Coding rate: 7
[INFO] setupRig : Power: 30 dBm
[INFO] setupRig : Sync: 0x12
[INFO] setupRig : CRC: 1
[INFO] setupRig : Header: explicit
[INFO] setupRig : Speed: 1255 bps
[INFO] setupRig : TOA (compressed): 0.24 sec
[INFO] setupRig : TOA (uncompressed): 0.41 sec
[INFO] setupRig : Min level: -129.63 dBm
**[ERROR] setupRig : Radio start error: -13**
[INFO] setupRig : Using SX126X module
**E (103) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first**
[INFO] setupRig : LoRa initialized
[INFO] setupBt : BT init[INFO] rigTask : Radio task started
 loraprs
[INFO] setupBt : BT initialized
[TRACE] onSerialRx : � c0
[TRACE] onSerialRx :  0
[TRACE] onRigTxBegin : onRigTxBegin
[TRACE] onSerialRx : � 82
[...]
```

PINOUT:
```
//in pins_arduino.h
//TX = 1;
//RX = 3;
//SDA = 21;
//SCL = 22;
//SS    = 5;
//MOSI  = 23;
//MISO  = 19;
//SCK   = 18;

//in config.h:
#define CFG_LORA_PIN_SS         5
#define CFG_LORA_PIN_RST        14
#define CFG_LORA_PIN_A          33 //DIO1
#define CFG_LORA_PIN_B          39 //BUSY - SVN 
#define CFG_LORA_PIN_RXEN       2
#define CFG_LORA_PIN_TXEN       4
#define BUILTIN_LED             12
```
Board: AZ-Delivery (esp32-wroom-32), ESP32 Dev module (in Arduino IDE from Board Manager), 
Radio module: E22 400M30S

No errors while compiling:

```
C:\Users\does_not_matter\Documents\Arduino\esp32_loraprs\esp32_loraprs.ino:11:48: note: #pragma message: Using default built-in config
 #pragma message("Using default built-in config")
                                                ^
C:\Users\does_not_matter\Documents\Arduino\esp32_loraprs\esp32_loraprs.ino:16:45: note: #pragma message: Configured for client mode
 #pragma message("Configured for client mode")
                                             ^
C:\Users\does_not_matter\Documents\Arduino\esp32_loraprs\loraprs_service.cpp: In member function 'void LoraPrs::Service::setupRig(long int, long int, int, int, int, int, int, bool)':
C:\Users\does_not_matter\Documents\Arduino\esp32_loraprs\loraprs_service.cpp:226:35: note: #pragma message: Using SX126X
     #pragma message("Using SX126X")
```

sh123, 2022-12-09:
https://jgromes.github.io/RadioLib/group__status__codes.html
#define | RADIOLIB_ERR_INVALID_OUTPUT_POWER   (-13)

yooshky, 2022-12-09:
OK, I see. 
I forgot the PA in the module. I changed the power to 20. Thanx.

Does this error matter?
```E (103) gpio: gpio_isr_handler_remove(480): GPIO isr service is not installed, call gpio_install_isr_service() first```

sh123, 2022-12-09:
I think it is not fatal, more like freertos warning.

closed 2022-12-09 by sh123

sh123, 2022-12-10:
I have the same message from FreeRTOS and it works fine.

yooshky, 2022-12-10:
GPIO14 ESP32 is connected to NRST module E22 400M30S
Pin permanently high.
I added 100nF and 100uF VCC-GND capacitors too

config.h:
`#define CFG_LORA_PIN_RST        14`

yooshky, 2022-12-11:
The examples from RadioLib works fine. My pinout is ok. 
The transmitter works when I manually set the TXN and RXN pins - ```digitalWrite(27, HIGH); digitalWrite(26, LOW);``` etc.; 
and automatically in RadioLib - ```radio.setRfSwitchPins(27, 26);``` 
I send my APRS position manually 
```int state = radio.transmit("SP9xxx-5>SP9xxx,WIDE2-2:!50xx.xxN/019xx.xxE>TEST LORA /73");``` 
I am received correctly. 
I haven't checked the interrupts yet, I had no time.

sh123, 2022-12-11:
If pinouts are correct and transmission does not work then I need more information, logs from console, information about your KISS client, have you tried APRSDroid?

yooshky, 2022-12-11:
Yes, I was using APRSDroid in KISS mode over BT with TRACE logging level. Position set manually. Something like that: 
```
[TRACE] onSerialRx : S 53
[TRACE] onSerialRx : P 50
[TRACE] onSerialRx : 9 39
[TRACE] onSerialRx : I 49
[TRACE] onSerialRx : N 4e
[TRACE] onSerialRx : Z 5a
[TRACE] onSerialRx : - 2d
[TRACE] onSerialRx : 4 34
[TRACE] onSerialRx : > 3e
[TRACE] onSerialRx : A 41
[TRACE] onSerialRx : P 50
[TRACE] onSerialRx : D 44
[TRACE] onSerialRx : R 52
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : 6 36
[TRACE] onSerialRx : , 2c
[TRACE] onSerialRx : W 57
[TRACE] onSerialRx : I 49
[TRACE] onSerialRx : D 44
[TRACE] onSerialRx : E 45
[TRACE] onSerialRx : 2 32
[TRACE] onSerialRx : - 2d
[TRACE] onSerialRx : 2 32
[TRACE] onSerialRx : : 3a
[TRACE] onSerialRx : = 3d
[TRACE] onSerialRx : 5 35
[TRACE] onSerialRx : 0 30
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : 7 37
[TRACE] onSerialRx : . 2e
[TRACE] onSerialRx : 5 35
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : N 4e
[TRACE] onSerialRx : / 2f
[TRACE] onSerialRx : 0 30
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : 9 39
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : 0 30
[TRACE] onSerialRx : . 2e
[TRACE] onSerialRx : 8 38
[TRACE] onSerialRx : 4 34
[TRACE] onSerialRx : E 45
[TRACE] onSerialRx : > 3e
[TRACE] onSerialRx :   20
[TRACE] onSerialRx : 
 a
[TRACE] onSerialRx : S 53
[TRACE] onSerialRx : P 50
[TRACE] onSerialRx : 9 39
[TRACE] onSerialRx : I 49
[TRACE] onSerialRx : N 4e
[TRACE] onSerialRx : Z 5a
[TRACE] onSerialRx : - 2d
[TRACE] onSerialRx : 4 34
[TRACE] onSerialRx : > 3e
[TRACE] onSerialRx : A 41
[TRACE] onSerialRx : P 50
[TRACE] onSerialRx : D 44
[TRACE] onSerialRx : R 52
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : 6 36
[TRACE] onSerialRx : , 2c
[TRACE] onSerialRx : W 57
[TRACE] onSerialRx : I 49
[TRACE] onSerialRx : D 44
[TRACE] onSerialRx : E 45
[TRACE] onSerialRx : 2 32
[TRACE] onSerialRx : - 2d
[TRACE] onSerialRx : 2 32
[TRACE] onSerialRx : : 3a
[TRACE] onSerialRx : = 3d
[TRACE] onSerialRx : 5 35
[TRACE] onSerialRx : 0 30
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : 7 37
[TRACE] onSerialRx : . 2e
[TRACE] onSerialRx : 5 35
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : N 4e
[TRACE] onSerialRx : / 2f
[TRACE] onSerialRx : 0 30
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : 9 39
[TRACE] onSerialRx : 1 31
[TRACE] onSerialRx : 0 30
[TRACE] onSerialRx : . 2e
[TRACE] onSerialRx : 8 38
[TRACE] onSerialRx : 4 34
[TRACE] onSerialRx : E 45
[TRACE] onSerialRx : > 3e
[TRACE] onSerialRx :   20
[TRACE] onSerialRx : 
 a
```
Nothing more after sending position in APRSDroid.
The waterfall on my RSP1 is empty. Pin RXEN permanently HIGH, TXEN permanently LOW.
This is a fragment of old logs that I left in the console. I'll do it again later and post the logs.

sh123, 2022-12-11:
Is it text based mode or ax.25? I see raw text

yooshky, 2022-12-11:
Indeed, text. I've tried different formats. I'll check again soon.

yooshky, 2022-12-11:
It works! That's the reason `#define CFG_TEXT_PACKETS      true `
Otherwise the APRSDroid does not work.

The receiver works great. But it sends frames like this:
```
2022-12-11 10:44:03 CET: INZ-4>APDR16,WIDE2-2,QA1M3,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 10:45:34 CET: INZ-4>APDR16,WIDE2-2,QA1M2,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 10:47:19 CET: INZ-4>APDR16,WIDE2-2,QA1M2,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 10:49:04 CET: INZ-4>APDR16,WIDE2-2,QA1M3,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 10:50:48 CET: INZ-4>APDR16,WIDE2-2,QA1M3,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 10:52:33 CET: INZ-4>APDR16,WIDE2-2,QA1M2,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 10:54:18 CET: INZ-4>APDR16,WIDE2-2,QA1M2,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 10:56:03 CET: INZ-4>APDR16,WIDE2-2,Q00L9,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 11:01:27 CET: INZ-4>APDR16,WIDE2-2,Q1492,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 11:02:25 CET: INZ-4>APDR16,WIDE2-2,Q1491,qAR,SP9QLJ-3:=5017.53N/01910.74E>
2022-12-11 11:02:39 CET: INZ-4>APDR16,WIDE2-2,Q1392,qAR,SP9QLJ-3:=5017.54N/01910.83E>
```
[https://aprs.fi/?c=raw&call=INZ-4](url)

My correct sign: SP9INZ. 
Same thing happens when i sending from RadioLib example `radio.transmit("SP9INZ-4>APDR16,WIDE2-2:!5017.51N/01910.84E>"); `
The command `radio.transmit("    SP9INZ-4>APDR16,WIDE2-2:!5017.51N/01910.84E>");` (4 extra spaces before the sign) gives the correct effect.

Thank you very much for your time

sh123, 2022-12-11:
`CFG_TEXT_PACKETS` `true` enables text based mode, which is compatible with https://github.com/lora-aprs/, they use proprietary protocol with 3 bytes prefix. To utilize it APRSDroid should use TNC2 text based mode. Myself I use only ax.25, because all open source tools use it and AFSK APRS making it compatible with them.

yooshky, 2022-12-11:
I have APRSDroid in TNC2 mode right now.
Otherwise don't work.

sh123, 2022-12-11:
If you set CFG_TEXT_PACKETS to false and use TNC KISS mode on APRSDroid then you need to receive it on the tracker which supports AX.25.

yooshky, 2022-12-11:
With this setting, ESP32 does not forward received data to APRSDroid (KISS). Sent data from APRSDroid in KISS mode are not decoded by gates and repeaters. I guess that's where my problem lies. APRSDroid and ESP32_lora don't understand each other (AX.25 and ASCII?)
```[TRACE] onRigTxBegin : onRigTxBegin<\n>
[TRACE] onSerialRx : â€š 82<\n>
[TRACE] onRigTx : â€š 82<\n>
[TRACE] onSerialRx : Â  a0<\n>
[TRACE] onRigTx : Â  a0<\n>
[TRACE] onSerialRx : Â 88<\n>
[TRACE] onRigTx : Â 88<\n>
[TRACE] onSerialRx : Â¤ a4<\n>
[TRACE] onRigTx : Â¤ a4<\n>
[TRACE] onSerialRx : b 62<\n>
[TRACE] onRigTx : b 62<\n>
[TRACE] onSerialRx : l 6c<\n>
[TRACE] onRigTx : l 6c<\n>
[TRACE] onSerialRx : Ĺ• e0<\n>
[TRACE] onRigTx : Ĺ• e0<\n>
[TRACE] onSerialRx : Â¦ a6<\n>
[TRACE] onRigTx : Â¦ a6<\n>
[TRACE] onSerialRx : Â  a0<\n>
[TRACE] onRigTx : Â  a0<\n>
[TRACE] onSerialRx : r 72<\n>
[TRACE] onRigTx : r 72<\n>
[TRACE] onSerialRx : â€™ 92<\n>
[TRACE] onRigTx : â€™ 92<\n>
[TRACE] onSerialRx : Ĺ› 9c<\n>
[TRACE] onRigTx : Ĺ› 9c<\n>
[TRACE] onSerialRx : Â´ b4<\n>
[TRACE] onRigTx : Â´ b4<\n>
[TRACE] onSerialRx : h 68<\n>
[TRACE] onRigTx : h 68<\n>
[TRACE] onSerialRx : Â® ae<\n>
[TRACE] onRigTx : Â® ae<\n>
[TRACE] onSerialRx : â€™ 92<\n>
[TRACE] onRigTx : â€™ 92<\n>
[TRACE] onSerialRx : Â 88<\n>
[TRACE] onRigTx : Â 88<\n>
[TRACE] onSerialRx : Ĺ  8a<\n>
[TRACE] onRigTx : Ĺ  8a<\n>
[TRACE] onSerialRx : d 64<\n>
[TRACE] onRigTx : d 64<\n>
[TRACE] onSerialRx : @ 40<\n>
[TRACE] onRigTx : @ 40<\n>
[TRACE] onSerialRx : e 65<\n>
[TRACE] onRigTx : e 65<\n>
[TRACE] onSerialRx : <3> 3<\n>
[TRACE] onRigTx : <3> 3<\n>
[TRACE] onSerialRx : Ä‘ f0<\n>
[TRACE] onRigTx : Ä‘ f0<\n>
[TRACE] onSerialRx : = 3d<\n>
[TRACE] onRigTx : = 3d<\n>
[TRACE] onSerialRx : 5 35<\n>
[TRACE] onRigTx : 5 35<\n>
[TRACE] onSerialRx : 0 30<\n>
[TRACE] onRigTx : 0 30<\n>
[TRACE] onSerialRx : 1 31<\n>
[TRACE] onRigTx : 1 31<\n>
[TRACE] onSerialRx : 7 37<\n>
[TRACE] onRigTx : 7 37<\n>
[TRACE] onSerialRx : . 2e<\n>
[TRACE] onRigTx : . 2e<\n>
[TRACE] onSerialRx : 5 35<\n>
[TRACE] onRigTx : 5 35<\n>
[TRACE] onSerialRx : 3 33<\n>
[TRACE] onRigTx : 3 33<\n>
[TRACE] onSerialRx : N 4e<\n>
[TRACE] onRigTx : N 4e<\n>
[TRACE] onSerialRx : / 2f<\n>
[TRACE] onRigTx : / 2f<\n>
[TRACE] onSerialRx : 0 30<\n>
[TRACE] onRigTx : 0 30<\n>
[TRACE] onSerialRx : 1 31<\n>
[TRACE] onRigTx : 1 31<\n>
[TRACE] onSerialRx : 9 39<\n>
[TRACE] onRigTx : 9 39<\n>
[TRACE] onSerialRx : 1 31<\n>
[TRACE] onRigTx : 1 31<\n>
[TRACE] onSerialRx : 0 30<\n>
[TRACE] onRigTx : 0 30<\n>
[TRACE] onSerialRx : . 2e<\n>
[TRACE] onRigTx : . 2e<\n>
[TRACE] onSerialRx : 9 39<\n>
[TRACE] onRigTx : 9 39<\n>
[TRACE] onSerialRx : 1 31<\n>
[TRACE] onRigTx : 1 31<\n>
[TRACE] onSerialRx : E 45<\n>
[TRACE] onRigTx : E 45<\n>
[TRACE] onSerialRx : > 3e<\n>
[TRACE] onRigTx : > 3e<\n>[
TRACE] onSerialRx :   20<\n>
[TRACE] onRigTx :   20<\n>
[TRACE] onSerialRx : Ĺ” c0<\n>
[TRACE] onRigTxEnd : onRigTxEnd 44<\n>
[TRACE] onRigTaskTxPacket : onRigTaskTxPacket 44<\n>
[TRACE] onRigTaskRxPacket : onRigTaskRxPacket 93<\n>```

yooshky, 2022-12-11:
At this point there is probably no chance to use APRSDroid with your software. Thank you very much for your help and time and support.

sh123, 2022-12-11:
Which gates are you using? Are they ax.25 gates or lora-aprs gates?

yooshky, 2022-12-11:
When i set `CFG_TEXT_PACKETS` to `false` and TNC2 in APRSDroid the effect is like this:
```
2022-12-11 10:44:03 CET: [INZ-4](https://aprs.fi/?c=raw&limit=&call=INZ-4)>APDR16,WIDE2-2,QA1M3,qAR,[SP9QLJ-3](https://aprs.fi/?c=raw&limit=&call=SP9QLJ-3):=5017.53N/01910.74E>
2022-12-11 10:45:34 CET: [INZ-4](https://aprs.fi/?c=raw&limit=&call=INZ-4)>APDR16,WIDE2-2,QA1M2,qAR,[SP9QLJ-3](https://aprs.fi/?c=raw&limit=&call=SP9QLJ-3):=5017.53N/01910.74E>
2022-12-11 10:47:19 CET: [INZ-4](https://aprs.fi/?c=raw&limit=&call=INZ-4)>APDR16,WIDE2-2,QA1M2,qAR,[SP9QLJ-3](https://aprs.fi/?c=raw&limit=&call=SP9QLJ-3):=5017.53N/01910.74E>
2022-12-11 10:49:04 CET: [INZ-4](https://aprs.fi/?c=raw&limit=&call=INZ-4)>APDR16,WIDE2-2,QA1M3,qAR,[SP9QLJ-3](https://aprs.fi/?c=raw&limit=&call=SP9QLJ-3):=5017.53N/01910.74E>
2022-12-11 10:50:48 CET: [INZ-4](https://aprs.fi/?c=raw&limit=&call=INZ-4)>APDR16,WIDE2-2,QA1M3,qAR,[SP9QLJ-3](https://aprs.fi/?c=raw&limit=&call=SP9QLJ-3):=5017.53N/01910.74E>
2022-12-11 10:52:33 CET: [INZ-4](https://aprs.fi/?c=raw&limit=&call=INZ-4)>APDR16,WIDE2-2,QA1M2,qAR,[SP9QLJ-3](https://aprs.fi/?c=raw&limit=&call=SP9QLJ-3):=5017.53N/01910.74E>
```
cuts off my sign (SP9INZ).
Software in my area and throughout Europe is compatible with [https://github.com/lora-aprs/LoRa_APRS_iGate](url)
I use this for SX1278 (e.g. RFM9x).

Also works: [https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS](url),

sh123, 2022-12-11:
Added change, pull from master, by default it will work without 3 byte prefix, if this interoperability is needed then there is a key `CFG_TEXT_PACKETS_PREFIX` which needs to be set to `true`, but default it is `false`.

yooshky, 2022-12-11:
Still cuts sign
`2022-12-11 17:40:38 CET: INZ-4>APDR16,WIDE2-2,qAR,SQ9IWE-2:=5017.52N/01910.84E> `
[https://aprs.fi/#!call=a%2FINZ-4&timerange=3600&tail=3600](url)

Raw text via LoRa:
```
int state = radio.transmit("SP9INZ-5>SP9INZ,WIDE1-1,WIDE2-2:!5017.51N/01910.84E>TEST, LORA 1W tracker /73"); 
```
gives the same effect.

This:
```
int state = radio.transmit("    SP9INZ-5>SP9INZ,WIDE1-1,WIDE2-2:!5017.51N/01910.84E>TEST, LORA 1W tracker /73"); 
```
causes the frame to be received correctly.

sh123, 2022-12-11:
https://github.com/oe3cjb/TTGO-T-Beam-LoRa-APRS/blob/master/lib/BG_RF95/BG_RF95.cpp

```
// Skip the 4 headers that are at the beginning of the rxBuf
	if (*len > _bufLen-BG_RF95_HEADER_LEN)
	    *len = _bufLen-(BG_RF95_HEADER_LEN-1);
```

Ok, so it drops 4 bytes at the beginning... I'll add that

yooshky, 2022-12-11:
Receiving the work is nice 
https://nas.yooshky.pl/share.cgi/Screenshot_20221211_180413_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=Screenshot_20221211_180413_APRSdroid.jpg&openfolder=normal&ep=

yooshky, 2022-12-11:
No, in a moment

sh123, 2022-12-11:
I've added change, need to set `CFG_TEXT_PACKETS_4` to `true` then it will add 4 bytes and drop 4 byte when receiving.

yooshky, 2022-12-11:
APRSDroid received frames well: https://nas.yooshky.pl/share.cgi/Screenshot_20221211_180413_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=Screenshot_20221211_180413_APRSdroid.jpg&openfolder=normal&ep=

yooshky, 2022-12-11:
Still cuts off my sign (3 bytes) and in received frames too (1 byte)
```
2022-12-11 17:47:32 CET: INZ-4>APDR16,WIDE2-2,qAR,SQ9IWE-2:=5017.52N/01910.84E> 
2022-12-11 17:49:17 CET: INZ-4>APDR16,WIDE2-2,Q1491,qAR,SP9QLJ-3:=5017.52N/01910.84E>
2022-12-11 17:49:17 CET: INZ-4>APDR16,WIDE2-2,qAR,SQ9IWE-2:=5017.52N/01910.84E> 
2022-12-11 17:50:55 CET: INZ-4>APDR16,WIDE2-2,QA4M6,qAR,SP9QLJ-2:=5017.52N/01910.84E>
```
config:
```
#define CFG_TEXT_PACKETS      true   // enable aprs text packets instead of binary for interoperability with other projects (disables KISS + AX.25!)
#define CFG_TEXT_PACKETS_3    false   // true - enable aprs-lora 3 byte prefix '<', 0xff, 0x01
#define CFG_TEXT_PACKETS_4    true   // true - enable oe3cjb 4 byte prefix
```
APRSDroid screen: https://nas.yooshky.pl/share.cgi/Screenshot_20221211_183135_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=Screenshot_20221211_183135_APRSdroid.jpg&openfolder=normal&ep=

yooshky, 2022-12-11:
Do not drop anything in the received frames. It was good

sh123, 2022-12-11:
Then just set `CFG_TEXT_PACKETS_4` to `false`

yooshky, 2022-12-11:
> 

> Then just set `CFG_TEXT_PACKETS_4` to `false`

https://nas.yooshky.pl/share.cgi/Screenshot_20221211_185543_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=Screenshot_20221211_185543_APRSdroid.jpg&openfolder=normal&ep=

sh123, 2022-12-11:
From the '<' prefix it looks like you need to set `CFG_TEXT_PACKETS_3` to `true` instead.

yooshky, 2022-12-11:
It does not add bytes in sent frames

sh123, 2022-12-11:
You want to add bytes to sent frames, but not drop for received? I do not understand this protocol.

yooshky, 2022-12-11:
> You want to add bytes to sent frames, but not drop for received? I do not understand this protocol.

Yes.  
https://nas.yooshky.pl/share.cgi/InkedScreenshot_20221211_191346_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=InkedScreenshot_20221211_191346_APRSdroid.jpg&openfolder=normal&ep=

sh123, 2022-12-11:
try from master

yooshky, 2022-12-11:
Now it adds bytes to received, not added to sent
https://nas.yooshky.pl/share.cgi/InkedScreenshot_20221211_193634_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=InkedScreenshot_20221211_193634_APRSdroid.jpg&openfolder=normal&ep=

sh123, 2022-12-11:
So, it should add 4 bytes into sent, but drop 3 in received?

yooshky, 2022-12-11:
should add 3 bytes to sent but drop 0 to received :)

sh123, 2022-12-11:
It was like that with latest change

yooshky, 2022-12-11:
Unfortunately it does not work. Does not add bytes to the sent frames, there are additional 3 bytes in the received ones. Look: https://nas.yooshky.pl/share.cgi/InkedScreenshot_20221211_193634_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=InkedScreenshot_20221211_193634_APRSdroid.jpg&openfolder=normal&ep=

yooshky, 2022-12-12:
```
#define CFG_TEXT_PACKETS_3    false
#define CFG_TEXT_PACKETS_4    true 
```
effect:  https://nas.yooshky.pl/share.cgi/InkedScreenshot_20221212_004859_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=InkedScreenshot_20221212_004859_APRSdroid.jpg&openfolder=normal&ep=

```
#define CFG_TEXT_PACKETS_3    true 
#define CFG_TEXT_PACKETS_4    true 
```
effect: https://nas.yooshky.pl/share.cgi/InkedScreenshot_20221212_011031_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=InkedScreenshot_20221212_011031_APRSdroid.jpg&openfolder=normal&ep=

```
#define CFG_TEXT_PACKETS_3    true
#define CFG_TEXT_PACKETS_4    false 
```
effect: https://nas.yooshky.pl/share.cgi/InkedScreenshot_20221212_012241_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=InkedScreenshot_20221212_012241_APRSdroid.jpg&openfolder=normal&ep=

```
#define CFG_TEXT_PACKETS_3    false
#define CFG_TEXT_PACKETS_4    false 
```
effect: https://nas.yooshky.pl/share.cgi/InkedScreenshot_20221212_013031_APRSdroid.jpg?ssid=02CWjJY&fid=02CWjJY&path=%2F&filename=InkedScreenshot_20221212_013031_APRSdroid.jpg&openfolder=normal&ep=

Green - sent by me
Blue/gray - received

yooshky, 2022-12-13:
First attempts:
[image]

I wrote a short program - GPS (UART) + NMEA parser and sending via RadioLib.
Unfortunately, I still can't use your software.

yooshky, 2022-12-13:
Gateways in my area use this: https://github.com/SQ9MDD/TTGO-T-Beam-LoRa-APRS and/or this: https://github.com/dl9sau/TTGO-T-Beam-LoRa-APRS **as TNC.** 

I use: https://github.com/lora-aprs/LoRa_APRS_iGate and https://github.com/lora-aprs/LoRa_APRS_Tracker.
I have no compatibility issues.

When I send via LoRa, the text _"SP9INZ-5>TEST,WIDE2-2...."_ via `radio.transmit()` from RadioLib, they receive with the first 3 bytes truncated _"INZ-5>TEST...."_.

The same happens when I send from APRSDroid through your TNC software (TNC2).

When I send via LoRa, the text etc. _"123SP9INZ-5>TEST,WIDE2-2...."_ via `radio.transmit()` from RadioLib, they receive corectly (without "123"). Iin my working simple tracker i add `unsigned char byte_tab[] = { 0x3c, 0xff, 0x01 };` and it works too.

You should insert 3 bytes in frames going out via LoRa. Do not change anything in received.

Sorry for my English. It's not my native language

sh123, 2022-12-13:
If you set CFG_TEXT_PACKETS and CFG_TEXT_PACKETS_3 to true it appends 0x3c, 0xff, 0x01 and expects these 3 bytes when receiving.

yooshky, 2022-12-13:
Then it works like this:
[image]

See this screenshot?
Do not add bytes in data received via LoRa.

sh123, 2022-12-13:
I do not add 3 bytes, I drop 3 bytes if they are equal to '<', 0xff, 0x01 and when these flags are set to true.

yooshky, 2022-12-13:
now it's like this:
[image]

efect:
[image]

yooshky, 2022-12-13:
You need to insert 3 bytes in frames coming out through LoRa. Cut 3 bytes if they exist in the received frame

sh123, 2022-12-13:
I do exactly that, add '<', 0xff, 0x01 and remove them if they exist, but I do not have aprs-lora tracker to test it, so only packet dump with trace debug level can help here.

yooshky, 2022-12-13:
Do you see my screenshots?

sh123, 2022-12-13:
I need packet dump from USB console on receiving and transmit with trace level to understand what does not work. Screenshots from aprs droid do not provide any clues, sorry.

yooshky, 2022-12-13:
I will send the logs from the console

sh123, 2022-12-13:
Try, please, from master with latest changes.

yooshky, 2022-12-13:
IT WORKS!!
You made me happy!
I will test if everything will be ok.
Now your software will be even more useful. 
I will be recommending to everyone for the E22 400M30S
I just have to go :)
https://aprs.fi/#!mt=roadmap&z=16&call=a%2FINZ-5%2Ca%2FSP9INZ-5&others=1&timerange=3600&tail=3600