sh123/esp32_loraprs #55: Collaboration with Codec2talkie

uw5elk, 2022-12-23:
1.Do I need to make any other changes to config.h (besides CFG_KISS_EXTENSIONS true) to use this project together with Codec2talkie.

 2. Is it possible to use the settings
 CFG_TEXT_PACKETS true
 CFG_TEXT_PACKETS_3 true
 when working with Codec2talkie?

sh123, 2022-12-23:
Codec2Talkie does not support TNC2 for APRS, only AX.25. You need to use APRSDroid TNC2 mode if you want to use with lora-arps iGate. Need to either add support for TNC2 into C2T or add support for packet translation (text <-> ax.25) into modem.

uw5elk, 2022-12-23:
That is, for C2T to work, set:
 CFG_TEXT_PACKETS false
 CFG_TEXT_PACKETS_3 false.
  Right?

sh123, 2022-12-23:
Yes, it will use KISS + AX.25 in this case, but lora-aprs iGate won't be able to receive AX.25 packets.

uw5elk, 2022-12-23:
So.  I'm just looking for a reason why C2T didn't work for me.  Thank you.

closed 2022-12-23 by sh123