sh123/esp32_loraprs #59: Digipeating in TNC2/text mode losing last character

fkaup, 2023-01-03:
Issue seems to be in loraprs_service.cpp lines 505ff: 

``` 
  // TNC2 text mode
  if (config_.EnableTextPackets) {
    String textPayload = payload.ToString();
    bytesWritten = textPayload.length() > (CfgMaxPacketSize - 1)? CfgMaxPacketSize : (textPayload.length() + 1);
    textPayload.getBytes(buf, bytesWritten);
    buf[bytesWritten-1] = '\0';
```

I updated above lines to work. Still, it should **not** work as that because we are reading more bytes than we should have in textPayload. Still, reducing `textPayload.getBytes(buf, bytesWritten-1);` again misses the last character of the message. I don´t know what is wrong here and how to properly fix it.

sh123, 2023-01-03:
String.length() returns string length without null terminator, so fix must be right. If we are reading that and place null into the last position, making last character to be lost. I'll add fix into master.

closed 2023-01-03 by sh123

fkaup, 2023-01-07:
it is just that we are now trying to read one character more than available from `textPayload` - which is then mitigated by the `getBytes` function limiting it again to `textPayload` size. So, it works but is not clean.