sh123/esp32_loraprs #60: Tracker and digirepeat

uw5elk, 2023-01-03:
Is it possible to work in tracker and digirepeat mode at the same time?

sh123, 2023-01-03:
Currently, it does not work as digirepeater when running in client mode. Usually, digirepeaters are stand alone devices located at high elevation point for larger coverage. For smaller coverage digirepeating mesh systems might be better solution, such as meshastic.

sh123, 2023-01-04:
Added in master, should digirepeat in client mode if digirepeater is enabled in config.

closed 2023-01-04 by sh123

uw5elk, 2023-01-04:
Thank you very much

uw5elk, 2023-01-05:
Compiling .pio\build\esp32dev\libe7c\RadioLib\modules\CC1101\CC1101.cpp.o
In file included from include/loraprs_service.h:12,
                 from src/loraprs_service.cpp:1:
include/config.h:30:68: note: #pragma message: LoRa pin definitions are not found, redefining...
 #pragma message("LoRa pin definitions are not found, redefining...")
                                                                    ^
include/config.h:51:58: note: #pragma message: BUILDIN_LED is not found, defining as 2
 #pragma message("BUILDIN_LED is not found, defining as 2")
                                                          ^
src/main.cpp:8:48: note: #pragma message: Using default built-in config
 #pragma message("Using default built-in config")
                                                ^
In file included from src/main.cpp:9:
include/config.h:30:68: note: #pragma message: LoRa pin definitions are not found, redefining...
 #pragma message("LoRa pin definitions are not found, redefining...")
                                                                    ^
include/config.h:51:58: note: #pragma message: BUILDIN_LED is not found, defining as 2
 #pragma message("BUILDIN_LED is not found, defining as 2")
                                                          ^
src/main.cpp:13:45: note: #pragma message: Configured for client mode
 #pragma message("Configured for client mode")
                                             ^
src/loraprs_service.cpp: In member function 'void LoraPrs::Service::setupRig(long int, long int, int, int, int, int, int, bool)':
src/loraprs_service.cpp:243:35: note: #pragma message: Using SX126X
     #pragma message("Using SX126X")
                                   ^
src/loraprs_service.cpp: In member function 'bool LoraPrs::Service::sendAx25PayloadToRig(const AX25::Payload&)':
src/loraprs_service.cpp:507:20: error: 'textPayloadLenght' was not declared in this scope
     bytesWritten = textPayloadLenght > CfgMaxPacketSize ? CfgMaxPacketSize : textPayloadLength;
                    ^~~~~~~~~~~~~~~~~
src/loraprs_service.cpp:507:20: note: suggested alternative: 'textPayloadLength'
     bytesWritten = textPayloadLenght > CfgMaxPacketSize ? CfgMaxPacketSize : textPayloadLength;
                    ^~~~~~~~~~~~~~~~~
                    textPayloadLength
*** [.pio\build\esp32dev\src\loraprs_service.cpp.o] Error 1

uw5elk, 2023-01-05:
1.0.10 gets up fine. Master - the above error.

sh123, 2023-01-05:
There was typo, fixed

uw5elk, 2023-01-05:
Thanks.  Now really great!