sh123/esp32_loraprs #62: CRC mismatch on ACKs

fkaup, 2023-01-20:
I'm trying to get direct messages working in TNC2/text mode. I'm using 2 boards running loraprs, one as iGate, the other as client connected to APRSdroid. Messages from APRS-IS to RF work. The ACK is generated in APRSdroid, sent to the client module and sent out but on the iGate I receive a CRC error. Please see the console output with LVL_TRACE attached. 

(Messages from the client to other stations work fine and the ACK is correctly received)

[ack-crc-error.txt](https://github.com/sh123/esp32_loraprs/files/10469282/ack-crc-error.txt)

fkaup, 2023-01-20:
Pair of messages attached. The ACK seems to be properly generated (44 chars, counted), but the iGate tries to read 62 chars. 

[ack-crc-errors.zip](https://github.com/sh123/esp32_loraprs/files/10470142/ack-crc-errors.zip)

sh123, 2023-01-21:
Not sure.. iGate starts receiving after transmitting, maybe got data from client before had a chance to switch to receive mode. I would check packets on a waterfall.

fkaup, 2023-01-21:
will try to check - but there's time between. Also well reproducible - haven´t seen as single ACK from the client properly received, so shouldn´t be any external traffic.

fkaup, 2023-01-24:
I have to correct myself. It was indeed interference from another digipeater (i.e. remote digipeater repeating packet to mobile station while mobile station transmits ACK). Testing with dummy loads on both ends results in the attached screenshots and logs. Same behavior with antennas on both sides. (Previously I had the mobile board closed off with a dummy load - which apparently reduced rx levels to a similar range as the remote digipeater)

``` 
[INFO] processIncomingRawPacketAsServer : DK2FK-1>APDR16,WIDE1-1::DK2FK-15 :ack0B}11
[INFO] processIncomingRawPacketAsServer : Packet sent to APRS-IS
```
[image]
[image]

closed 2023-01-24 by fkaup