sh123/esp32_loraprs #64: PCB holes for ESP 32 module is too narrow

Fotr0, 2023-03-18:
Hi! 

Can you please explain what type of ESP32 module is used for your PSB design? 

I have makes PSBs from Gerber from this folder

https://github.com/sh123/esp32_loraprs/tree/master/extras/schematics/gerber

...and ordered this ESP32 Wroom 32 - 38pins modules, like this 

https://github.com/sh123/esp32_loraprs/blob/master/extras/images/esp32_38pin.jpg

but the rows of contacts is too narrow (actually the module rows is wider).

[image]
[image]
[image]

sh123, 2023-03-18:
Yes, there are boards of different pin numbers 30 and 38 (these are configured with solder pads), also boards with different PCB width, the narrow is on the right, I also have some wider boards, had to modify pins a little bit to fit. 

[image]

Fotr0, 2023-03-18:
Thanks! Could you please take me url for correct module? Thanks!

sh123, 2023-03-18:
Search for NodeMCU-32S, for example

https://www.aliexpress.com/item/1005002875650002.html?spm=a2g0o.productlist.main.53.1ca87404Kk23sg&algo_pvid=a1ed667c-2394-40ec-a6f6-c7aeb8069ab4&algo_exp_id=a1ed667c-2394-40ec-a6f6-c7aeb8069ab4-26&pdp_ext_f=%7B%22sku_id%22%3A%2212000022575979770%22%7D&pdp_npi=3%40dis%21EUR%216.31%213.79%21%21%21%21%21%402122443916791649023534608d06e0%2112000022575979770%21sea%21EE%21720589101&curPageLogUid=Lf7LxxRVGqRW

Check for distance between board pinouts and module, they must be closer.

closed 2023-03-18 by Fotr0

sh123, 2023-03-18:
@Fotr0 , added separate pcb and gerber files for 25mm wide esp32 socket boards, with _wide suffix in file names, did not test.