sh123/esp32_loraprs #67: lora32 v 2.1

EA2CW, 2023-08-09:
Hello, can this firmware be used on the lora32 v2.1 1.6 card? 
I would love to be able to use this cheap card as KISS TNC for mobile phones.
Thank you!

uw5elk, 2023-08-09:
https://github.com/dl9sau/TTGO-T-Beam-LoRa-APRS

sh123, 2023-08-09:
@EA2CW , in theory it can be used with any esp32 and sx127x/sx127x lora modules, you just need to define pinouts and other parameters yourself as I do not have this board hence no pre-defined config available.

EA2CW, 2023-08-09:
Thank you I've already tried it, but there is a problem between http server and bluetooth. You cannot have both activated at the same time. But you need the web server as the configuration must be made thru it. So you must compile / upload the firmware twice. One for config and second for enable bluetooth. 
Not very easy for normal people (and even harder to explain it)
The firmware has 2 config files (that would be right), but it simply ignores them
Thank you anyway

sq2mo, 2023-08-21:
@EA2CW There is an option in DL9SAU's firmware which turns on BT initially and switches to WiFi if no BT client has connected in 60 seconds.

EA2CW, 2023-08-21:
I know, and I'm using it now, but that was not my question, thank you!

closed 2023-08-22 by sh123