sh123/esp32_loraprs #69: Help with ax25

richonguzman, 2023-11-21:
Is there a way to contact you to learn to decode and encode ax25 packets to avoid copying all without understanding what you did? 

This is for a Lora Tracker in developing

sh123, 2023-11-21:
I doubt I can explain it better than it is already explained @ http://www.ax25.net/

richonguzman, 2023-11-21:
Thanks for the links (I've been reading all of them but somehow always I got lost in the way the ax25 should be builded with the APRS-packet)

Maybe is a language translation issue for me .

But I still can build a small map as how to get into the encoded ax25

Should I transform APRS packet into kiss and then this kiss into ax25 encoding part by part?

sh123, 2023-11-21:
This is the format of the AX.25 UI packet, APRS data is sent inside information field, flag and CRC checksum are not used, they are part of underneath HDLC protocol, which is used in causal AFSK1200 APRS.

So, if you have APRS text packet, let's say
```
FROM>TO,DIGI1,DIGI2,DIGIx:info
```
Then you need to place FROM to source address, TO to destination, DIGIx to digirepeater addresses and info into information field.

[image]

closed 2023-11-29 by sh123

DerHirschi, 2023-11-29:
U also have take about KISS Protocol Bit-Stuffing.

Sry, but i just found this Specs in German Language.
"Um zu verhindern, daß ein Flag innerhalb der Daten auftritt, wird das sogenannte Bit-Stuffing
verwendet.
Dabei fügt der Sender nach der Übertragung von jeweils fünf aufeinanderfolgenden Eins-Bits
ein Null-Bit in den seriellen Datenstrom ein. Hat der Empfänger also fünf aufeinanderfolgende
Eins-Bits empfangen, und folgt ein Null-Bit, wird dieses entfernt. Folgt ein weiteres Eins-Bit,
handelt es sich um ein Flag, falls darauf eine Null folgt. Eine Folge von mehr als sechs Eins-
Bits wird als Fehlercode interpretiert und der empfangene Block wird verworfen."

"To prevent a flag from occurring within the data, so-called bit stuffing is used.
is used.
In this process, the sender inserts a zero bit into the serial data stream after each transmission of five consecutive one bits.
a zero bit into the serial data stream. So if the receiver has received five consecutive
one bits followed by a zero bit, this is removed. If another one-bit follows,
is a flag if it is followed by a zero. A sequence of more than six one
bits is interpreted as an error code and the received block is discarded.

Translated with DeepL.com (free version)"
https://github.com/DerHirschi/AX25_POPT/blob/dev/doc/specs/KISS-AX25.pdf

sh123, 2023-11-29:
Bit stuffing is needed for HDLC, why do you need HDLC over LoRa, HDLC might be needed if using FSK direct mode with the modem.

richonguzman, 2023-11-29:
As far as I could find, for Lora APRS and BLE , only ax25 frame , encapsulated in Kiss bytes is needed , right???

DerHirschi, 2023-11-29:
Sure, for LoRa you don't need a preamble/postamble or a checksum.
On KISS over TCP like in Direwolf you don't get the pre/post and checksum either.

But for backend software like KissAttach I need an AX25 frame encapsulated in a KISS frame as Richonguzman said.
Sorry if there was a misunderstanding with my comment. I think I was wrong about the bit stuffing or meant something wrong.

I wanted to point out the following:

from: https://github.com/ampledata/kiss
Escape special codes, per KISS spec.
"If the FEND or FESC codes appear in the data to be transferred, they
need to be escaped. The FEND code is then sent as FESC, TFEND and the
FESC is then sent as FESC, TFESC."
- http://en.wikipedia.org/wiki/KISS_(TNC)#Description

FESC   = b'\xDB'
TFESC = b'\xDD'

Sorry for the confusion

richonguzman, 2023-11-29:
right now I have as a BLE frame (I validated the generated BLE frame on https://notblackmagic.com/bitsnpieces/ax.25/ and it works )

Fend Data 7bytes-Tocall 7bytes-callsing Control PID payload Fend but the aprs.fi app does not acknowledge nor receive the BLE frame.

7bytes-tocall and 7bytes-callsing = 6byte-address + 1 byteSSID

Am I missing something?

DerHirschi, 2023-11-29:
In the SSID Byte are just 4 Bit's for SSID ..

sh123, 2023-11-30:
For APRS-IS servers you need to extract AX.25 from KISS (inc. unescaping), then convert to text format and then send to APRS-IS.

richonguzman, 2023-11-30:
> 
> Fend Data 7bytes-Tocall 7bytes-callsing Control PID payload Fend 
> 

Success 1/2 way....  aprs.fi iOS app is receiving the BLE frame and showing it on screen 

the only last thing to get right is the SSID calculations ...  

I've seen to many different ways to do this that I don't know how to:
is it CRRSSSS0 sometimes? os C=0 for Sender/Callsign and 1 for Tocall?

should I only extract SSSS from this byte?

sh123, 2023-11-30:
Callsign is 7 bytes, to encode to binary you need to shift left each byte by 1 bit, to decode from binary you need to shift right. Space should be replaced with \0 and last bit indicates end of digirepeater list.

sh123, 2023-11-30:
https://notblackmagic.com/bitsnpieces/ax.25/