sh123/esp32_loraprs #72: Problem during start

tortyl, 2023-12-07:
Hello,

My setup:
ESP32
RA02 - SX1278

During startup I receive message:

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[INFO] setup : KISS extensions are disabled in TNC2 mode
[INFO] setupRig : Initializing LoRa
[INFO] setupRig : Frequency: 433775000 Hz
[INFO] setupRig : Bandwidth: 125000 Hz
[INFO] setupRig : Spreading: 12
[INFO] setupRig : Coding rate: 7
[INFO] setupRig : Power: 20 dBm
[INFO] setupRig : Sync: 0x12
[INFO] setupRig : CRC: 1
[INFO] setupRig : Header: explicit
[INFO] setupRig : Speed: 209 bps
[INFO] setupRig : TOA (compressed): ~ 1.42 sec
[INFO] setupRig : TOA (uncompressed): ~ 2.45 sec
[INFO] setupRig : Min level: -137.03 dBm
[INFO] printConfig : Version: LoRAPRS 1.0.15
[INFO] printConfig : Current mode: APRS-IS iGate
[INFO] printConfig : Using TNC2 text mode
[INFO] printConfig : UsbSerialEnable: no
[INFO] printConfig : EnableSignalReport: yes
[INFO] printConfig : EnablePersistentAprsConnection: yes
[INFO] printConfig : EnableRfToIs: yes
[INFO] printConfig : EnableIsToRf: no
[INFO] printConfig : EnableRepeater: no
[INFO] printConfig : EnableBeacon: yes
[   106][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[INFO] setupRig : Using SX127X module
[INFO] setupRig : LoRa initialized
[INFO] setupWifi : WIFI connecting to DomekTomek
[INFO] rigTask : Radio task started
[INFO] setupWifi : WIFI retrying 0
[INFO] setupWifi : WIFI retrying 1
[INFO] setupWifi : WIFI connected to DomekTomek
[INFO] setupWifi : IP address: 192.168.0.169
[INFO] reconnectAprsis : APRSIS connecting to rotate.aprs2.net
[INFO] reconnectAprsis : APRSIS connected
[INFO] reconnectAprsis : APRSIS logged in
[INFO] sendPeriodicBeacon : Periodic beacon is sent
[TRACE] onRigTxBegin : onRigTxBegin
[TRACE] onRigTaskStartTx : onRigTaskStartTx
[TRACE] onRigTx : < 3c
[TRACE] onRigTx : � ff
[TRACE] onRigTx : ␁ 1
[TRACE] onRigTx : S 53
[TRACE] onRigTx : Q 51
[TRACE] onRigTx : 6 36
[TRACE] onRigTx : S 53
[TRACE] onRigTx : L 4c
[TRACE] onRigTx : B 42
[TRACE] onRigTx : - 2d
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 0 30
[TRACE] onRigTx : > 3e
[TRACE] onRigTx : A 41
[TRACE] onRigTx : P 50
[TRACE] onRigTx : Z 5a
[TRACE] onRigTx : M 4d
[TRACE] onRigTx : D 44
[TRACE] onRigTx : M 4d
[TRACE] onRigTx : , 2c
[TRACE] onRigTx : W 57
[TRACE] onRigTx : I 49
[TRACE] onRigTx : D 44
[TRACE] onRigTx : E 45
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : - 2d
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : : 3a
[TRACE] onRigTx : ! 21
[TRACE] onRigTx : 5 35
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 0 30
[TRACE] onRigTx : 7 37
[TRACE] onRigTx : . 2e
[TRACE] onRigTx : 8 38
[TRACE] onRigTx : 4 34
[TRACE] onRigTx : N 4e
[TRACE] onRigTx : L 4c
[TRACE] onRigTx : 0 30
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 6 36
[TRACE] onRigTx : 4 34
[TRACE] onRigTx : 7 37
[TRACE] onRigTx : . 2e
[TRACE] onRigTx : 8 38
[TRACE] onRigTx : 2 32
[TRACE] onRigTx : E 45
[TRACE] onRigTx : a 61
[TRACE] onRigTx : L 4c
[TRACE] onRigTx : o 6f
[TRACE] onRigTx : R 52
[TRACE] onRigTx : A 41
[TRACE] onRigTx :   20
[TRACE] onRigTx : D 44
[TRACE] onRigTx : i 69
[TRACE] onRigTx : g 67
[TRACE] onRigTx : i 69
[TRACE] onRigTx :   20
[TRACE] onRigTx : i 69
[TRACE] onRigTx : G 47
[TRACE] onRigTx : a 61
[TRACE] onRigTx : t 74
[TRACE] onRigTx : e 65
[TRACE] onRigTx :   20
[TRACE] onRigTx : 4 34
[TRACE] onRigTx : 3 33
[TRACE] onRigTx : 3 33
[TRACE] onRigTx : . 2e
[TRACE] onRigTx : 7 37
[TRACE] onRigTx : 7 37
[TRACE] onRigTx : 5 35
[TRACE] onRigTx : M 4d
[TRACE] onRigTx : H 48
[TRACE] onRigTx : z 7a
[TRACE] onRigTx : / 2f
[TRACE] onRigTx : B 42
[TRACE] onRigTx : W 57
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 2 32
[TRACE] onRigTx : 5 35
[TRACE] onRigTx : / 2f
[TRACE] onRigTx : S 53
[TRACE] onRigTx : F 46
[TRACE] onRigTx : 1 31
[TRACE] onRigTx : 2 32
[TRACE] onRigTx : ␀ 0
[TRACE] onRigTxEnd : onRigTxEnd 86
[TRACE] onRigTaskTxPacket : onRigTaskTxPacket 86
[ERROR] onRigTaskTxPacket : TX error:  -5
[TRACE] onRigTaskStartRx : onRigTaskStartRx
[INFO] onAprsisDataAvailable : # aprsc 2.1.14-g5e22b37
[INFO] onAprsisDataAvailable : # logresp SQ6SLB-10 verified, server T2NANJING
[INFO] onAprsisDataAvailable : # aprsc 2.1.14-g5e22b37 7 Dec 2023 07:41:17 GMT T2NANJING 172.18.0.2:14580

Although the beacon was sent it receives an error: TX error: -5
Unfortunately, the device is not receiving anything from LORA and stops flashing the LED on the board.

Regards
Tomasz

sh123, 2023-12-07:
https://jgromes.github.io/RadioLib/group__status__codes.html

RADIOLIB_ERR_TX_TIMEOUT   (-5)   Timed out waiting for transmission finish.

tortyl, 2023-12-07:
> https://jgromes.github.io/RadioLib/group__status__codes.html
> 
> RADIOLIB_ERR_TX_TIMEOUT (-5) Timed out waiting for transmission finish.

What could be the cause?

sh123, 2023-12-07:
From https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX127x.cpp

I see that it sets RADIOLIB_ERR_TX_TIMEOUT, when there is no IRQ signal coming back from the module indicating transmission completion.

```
// wait for packet transmission or timeout
  start = this->mod->hal->micros();
  while(!this->mod->hal->digitalRead(this->mod->getIrq())) {
    this->mod->hal->yield();
    if(this->mod->hal->micros() - start > timeout) {
      finishTransmit();
      return(RADIOLIB_ERR_TX_TIMEOUT);
    }
  }
```
Might be misconfigured pin, this line broken or some trouble with the module. Did it work with older versions?

tortyl, 2023-12-07:
It works ! Wiring problem.
Thank you for the hint.